OrcaSlicer/OrcaSlicer #2702: OrcaSlicer calibration tools should not make unnecessary user profile changes

R4toLoko, 2023-11-12:
This isn't a feature request but I must fix logic that is wrong.

OrcaSlicer keeps changing the profile and that is wrong.
During a calibration process, you wanna perform it using the same profile used to print a model and not some random settings.
PA calibration makes these changes.

[image]

30mm/s is painful slow and creates a hell of resonance on my Voron.
A material behaves differently under different speed and different line heights.
I have to often hit Klipper emergency stop when I realized I forgot to hit "Reset settings" to use my profile instead.

[image]

DVSVIDEO, 2023-11-14:
I do not understand the problem.
During calibration it sets the values that are necessary to perform the calibration. These changes are indicated by an orange colour and a small circular arrow.
At the end of the calibration when I close the program, it asks me to save the changes. I choose "discard" and my profile remains the same.
And if I want to work without quitting, I press this little arrow: 
[image]
And everything goes back to the original.
Where is the problem here?

R4toLoko, 2023-11-15:
> I do not understand the problem. During calibration, it automatically sets the values needed for calibration. These changes are indicated by an orange colour and a small circular arrow. I guess it uses my profile because it only changes the ones that are needed for a successful test, everything else stays my own. At the end of the calibration, when I close the program, it asks me to save the changes. I choose "discard" and my profile remains unchanged. And if I want to work without quitting, I press this little arrow: [image] And everything goes back to the original. Where is the problem here?

It shouldn't happen in the first place and the reset does not work for everything.
An user is having the same problem with PA line and nope, you cannot reset the speed being forced to print 30mm/s

[image]

Take max vol as example, those changes makes perfectly sense. Only what is really needed change, nothing else.
It isn't hard to understand.

[image]

R4toLoko, 2023-11-15:
> I do not understand the problem. During calibration, it automatically sets the values needed for calibration. These changes are indicated by an orange colour and a small circular arrow. I guess it [...]

I use this slicer since its very first release, I know how it works :)
If you are happy printing 30mm/s, good for you.
It doesn't matter if the reset button is right there, it doesn't always work like explained above.

Printing quality does change according to speed, temp, accel, etc, calibrating a filament using the stock values but using your profile later on, will generate prints with different behaviors. That isn't new either.

DVSVIDEO, 2023-11-15:
Then something is not working correctly for you (in your software). I tried the PA test and it doesn't change anything for me. Nothing!
Then I also tried the max flowrate test. Here it changes a lot but with one button press everything is back to normal, or if you start a new project and select "Discard" everything is back to the original profile.

Evidence:
[image]

Start a new project:
[image]

New project at the right speeds: 
[image]

I still don't understand what the problem is?

SoftFever, 2023-11-16:
It's recommended to create a new project and don't save settings once the calibration is done, just

DVSVIDEO, 2023-11-18:
I don't think it has to be complicated. Make several copies of your profile (with numbered names).
 And if you accidentally overwrite one, there's always a backup. It's that simple.

Much simpler and only a few people feel the need, than having to change the whole program and its logical structure, which then has to be relearned and got used to by all people.
Not everything in the world can be foolproof...

HakunMatat4, 2023-11-29:
1. I don't think we are talking about the same problem
2. At no moment I said about saving profile, this isn't my first print, I know how the profiles management work!
3. I should have explained what I think it is a problem better

Why does PA wanna set my speed to nothing???
It should use the user profile speed.

[image]

I have to remember to reset it every time.
I believe that you all know or should know that **a corner printed at 30mm/s will behave completely different than 100mm/s**, it is just logic, there is nothing rocket science about that.

[image]

It was a copy/paste from Ellis template, you must insert your own settings which on OrcaSlicer side, the user profile is already there, just use it.

[image]

It looks like I am the only one seeing the problem so whatever, I am closing this.

closed 2023-11-29 by HakunMatat4