OrcaSlicer/OrcaSlicer #2703: No content in windows when running

gregersn, 2023-11-12:
### OrcaSlicer Version

1.8.0-rc2

### OS version

Fedora 39

### Printer

N/A

### How to reproduce

Starting the app-image gives blank windows. Both the getting started wizard and main window are white

```
❯ ./OrcaSlicer_Linux_V1.8.0-rc2.AppImage
[2023-11-12 10:18:38.786090] [0x00007f272f8b7e80] [trace]   Initializing StaticPrintConfigs
add font of HarmonyOS_Sans_SC_Bold returns 1
add font of HarmonyOS_Sans_SC_Regular returns 1
add font of NanumGothic-Regular returns 1
add font of NanumGothic-Bold returns 1

(orca-slicer:2905637): Gtk-CRITICAL **: 10:18:39.123: gtk_box_gadget_distribute: assertion 'size >= 0' failed in GtkScrollbar
loading network module at /home/greger/.config/OrcaSlicer/plugins/libbambu_networking.so
error, dlerror is /home/greger/.config/OrcaSlicer/plugins/libbambu_networking.so: cannot open shared object file: No such file or directory
after dlopen, network_module is (nil)
```

### Actual results

No interface

### Expected results

An interface

### Project file & Debug log uploads

[log.tar.gz](https://github.com/SoftFever/OrcaSlicer/files/13327617/log.tar.gz)

### Checklist of files to include

- [X] Log file
- [ ] Project file

Noisyfox, 2023-11-12:
Try run `export WEBKIT_DISABLE_COMPOSITING_MODE=1` then run the app.

gregersn, 2023-11-12:
> Try run `export WEBKIT_DISABLE_COMPOSITING_MODE=1` then run the app.

This does work, thank you

etrigan63, 2023-11-23:
`export WEBKIT_DISABLE_COMPOSITING_MODE=1` does not work for me.
[image]
CLI is littered with
```
[2023-11-23 00:08:47.770039] [0x00007f719c2fe8c0] [trace]   Initializing StaticPrintConfigs
add font of HarmonyOS_Sans_SC_Bold returns 1
add font of HarmonyOS_Sans_SC_Regular returns 1
add font of NanumGothic-Regular returns 1
add font of NanumGothic-Bold returns 1
loading network module at /home/guru/.config/OrcaSlicer/plugins/libbambu_networking.so
error, dlerror is /home/guru/.config/OrcaSlicer/plugins/libbambu_networking.so: cannot open shared object file: No such file or directory
after dlopen, network_module is (nil)

(orca-slicer:353229): GLib-GObject-CRITICAL **: 00:08:51.281: invalid cast from 'wxPizza' to 'GtkCellLayout'

(orca-slicer:353229): Gtk-CRITICAL **: 00:08:51.281: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed

(orca-slicer:353229): GLib-GObject-CRITICAL **: 00:08:51.282: invalid cast from 'wxPizza' to 'GtkCellLayout'

(orca-slicer:353229): Gtk-CRITICAL **: 00:08:51.282: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed

(orca-slicer:353229): GLib-GObject-CRITICAL **: 00:08:51.283: invalid cast from 'wxPizza' to 'GtkCellLayout'

(orca-slicer:353229): Gtk-CRITICAL **: 00:08:51.283: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed

(orca-slicer:353229): GLib-GObject-CRITICAL **: 00:08:51.283: invalid cast from 'wxPizza' to 'GtkCellLayout'

(orca-slicer:353229): Gtk-CRITICAL **: 00:08:51.283: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed

(orca-slicer:353229): GLib-GObject-CRITICAL **: 00:08:51.284: invalid cast from 'wxPizza' to 'GtkCellLayout'

(orca-slicer:353229): Gtk-CRITICAL **: 00:08:51.284: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed

(orca-slicer:353229): GLib-GObject-CRITICAL **: 00:08:51.285: invalid cast from 'wxPizza' to 'GtkCellLayout'

(orca-slicer:353229): Gtk-CRITICAL **: 00:08:51.285: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed

(orca-slicer:353229): GLib-GObject-CRITICAL **: 00:08:51.286: invalid cast from 'wxPizza' to 'GtkCellLayout'

(orca-slicer:353229): Gtk-CRITICAL **: 00:08:51.286: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed

(orca-slicer:353229): GLib-GObject-CRITICAL **: 00:08:51.286: invalid cast from 'wxPizza' to 'GtkCellLayout'

(orca-slicer:353229): Gtk-CRITICAL **: 00:08:51.286: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed

(orca-slicer:353229): GLib-GObject-CRITICAL **: 00:08:51.287: invalid cast from 'wxPizza' to 'GtkCellLayout'

(orca-slicer:353229): Gtk-CRITICAL **: 00:08:51.287: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed

(orca-slicer:353229): GLib-GObject-CRITICAL **: 00:08:51.288: invalid cast from 'wxPizza' to 'GtkCellLayout'

(orca-slicer:353229): Gtk-CRITICAL **: 00:08:51.288: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed

(orca-slicer:353229): Gtk-CRITICAL **: 00:08:51.349: gtk_window_resize: assertion 'height > 0' failed

(orca-slicer:353229): Gtk-CRITICAL **: 00:08:51.351: gtk_widget_set_size_request: assertion 'width >= -1' failed

(orca-slicer:353229): Gtk-CRITICAL **: 00:08:51.351: gtk_widget_set_size_request: assertion 'width >= -1' failed

(orca-slicer:353229): Gtk-CRITICAL **: 00:08:51.351: gtk_widget_set_size_request: assertion 'width >= -1' failed

(orca-slicer:353229): Gtk-CRITICAL **: 00:08:51.351: gtk_widget_set_size_request: assertion 'width >= -1' failed

(orca-slicer:353229): Gtk-CRITICAL **: 00:08:51.351: gtk_widget_set_size_request: assertion 'width >= -1' failed

(orca-slicer:353229): Gtk-CRITICAL **: 00:08:51.352: gtk_widget_set_size_request: assertion 'width >= -1' failed

(orca-slicer:353229): Gtk-CRITICAL **: 00:08:51.352: gtk_widget_set_size_request: assertion 'width >= -1' failed

(orca-slicer:353229): Gtk-CRITICAL **: 00:08:51.352: gtk_widget_set_size_request: assertion 'width >= -1' failed
[... 214 lines cut ...]

(orca-slicer:353229): GLib-GObject-CRITICAL **: 00:08:53.981: invalid cast from 'wxPizza' to 'GtkCellLayout'

(orca-slicer:353229): Gtk-CRITICAL **: 00:08:53.981: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed

(orca-slicer:353229): GLib-GObject-CRITICAL **: 00:08:53.981: invalid cast from 'wxPizza' to 'GtkCellLayout'

(orca-slicer:353229): Gtk-CRITICAL **: 00:08:53.981: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed

(orca-slicer:353229): GLib-GObject-CRITICAL **: 00:08:53.982: invalid cast from 'wxPizza' to 'GtkCellLayout'

(orca-slicer:353229): Gtk-CRITICAL **: 00:08:53.982: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed

(orca-slicer:353229): GLib-GObject-CRITICAL **: 00:08:53.995: invalid cast from 'wxPizza' to 'GtkCellLayout'

(orca-slicer:353229): Gtk-CRITICAL **: 00:08:53.995: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed

(orca-slicer:353229): GLib-GObject-CRITICAL **: 00:08:53.995: invalid cast from 'wxPizza' to 'GtkCellLayout'

(orca-slicer:353229): Gtk-CRITICAL **: 00:08:53.995: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed

** (orca-slicer:353229): CRITICAL **: 00:08:54.008: Cannot register URI scheme wxfs more than once

** (orca-slicer:353229): CRITICAL **: 00:08:54.008: Cannot register URI scheme memory more than once
```

Skaifer, 2023-12-18:
I have the same issue.  `export WEBKIT_DISABLE_COMPOSITING_MODE=1` does not help. Terminal output is about the same as @etrigan63. The app opens with wizard with blank body, there is nothing other than top panel categories on main window.

Skaifer, 2023-12-18:
It appears, that setting this variable but in a different way actually solves the issue and the app renders and slices normally now. By launching trough terminal this way `env WEBKIT_DISABLE_COMPOSITING_MODE=1 ./OrcaSlicer_Linux_V1.8.1.AppImage `, or in KDE you can right-click on desktop file and  `Edit application > Application > Environment Variables:` and insert `WEBKIT_DISABLE_COMPOSITING_MODE=1` into that field.

cduhard, 2023-12-19:
I am experiencing a similar thing. the webkit env variable does nothing. There is no UI for adding anything to the build plate. Just blank.

Kalitsune, 2024-01-04:
> `export WEBKIT_DISABLE_COMPOSITING_MODE=1` does not work for me. [...]

I got the same issue but I managed to fix it by doing a combinaison of `export WEBKIT_DISABLE_COMPOSITING_MODE=1 and the installation of the devel version of the package (`webkit2gtk4.0-devel` on fedora)

Kalitsune, 2024-01-04:
here's the .desktop file I ended up going for if someone is interested:
```cfg
[Desktop Entry]
Name=OrcaSlicer
Exec=sh -c "env WEBKIT_DISABLE_COMPOSITING_MODE=1 /path/to/OrcaSlicer_Linux_V1.8.1.AppImage %F"
Icon=/path/to/OrcaSlicer.png
Type=Application
Categories=Utility;
MimeType=model/stl;application/vnd.ms-3mfdocument;application/prs.wavefront-obj;application/x-amf;
Name[en]=OrcaSlicer
Name[en_US]=OrcaSlicer
Name[en_US.UTF-8]=OrcaSlicer
Hidden=false
NoDisplay=false
StartupNotify=false
Terminal=false
```

Killdozer666, 2024-01-13:
Thank you, this fixed it for me. I used it on both Bambu Studio and Orca Slicer.
I tried it on terminal first using:
```
Exec=env WEBKIT_DISABLE_COMPOSITING_MODE=1 /home/path/BambuStudio_linux_ubuntu_v01.08.03.89.AppImage %F
```

once tested, I created a App Launcher Menu entry, then adding the line of code to the .desktop file. It looks like this:

```
[Desktop Entry]
Name=Bambu Studio 1.8.3
Exec=/home/path/BambuStudio_linux_ubuntu_v01.08.03.89.AppImage
Comment=
Exec=env WEBKIT_DISABLE_COMPOSITING_MODE=1 /home/path/BambuStudio_linux_ubuntu_v01.08.03.89.AppImage %F
Terminal=false
PrefersNonDefaultGPU=true
Icon=/home/path/BambuStudio.png
Type=Application
Hidden=false
NoDisplay=false
StartupNotify=false
Terminal=false
```
Now Bambu Studio  launches fast, no errors, and no annoying "X Stoped Responding" popup error at launch.  Everything Renders as it should.

**But for Orca Slicer, I still get the "X Stoped Responding" popup error at launch. and when I do it on terminal I still see all the "Critical Error" freakout you see above that other have posted.  The app renders now and it works fine.  I don't even have webkit installed currently (I uninstalled for testing), but the "critial errors" are still there.**

Thank you for your assistance.

Will keep testing.

elegant-dragon-whisperer, 2024-03-13:
> Thank you, this fixed it for me. I used it on both Bambu Studio and Orca Slicer. I tried it on terminal first using:
> 
> ```
> Exec=env WEBKIT_DISABLE_COMPOSITING_MODE=1 /home/path/BambuStudio_linux_ubuntu_v01.08.03.89.AppImage %F
> ```
> 
> once tested, I created a App Launcher Menu entry, then adding the line of code to the .desktop file. It looks like this:
> 
> ```
> [Desktop Entry]
> Name=Bambu Studio 1.8.3
> Exec=/home/path/BambuStudio_linux_ubuntu_v01.08.03.89.AppImage
> Comment=
> Exec=env WEBKIT_DISABLE_COMPOSITING_MODE=1 /home/path/BambuStudio_linux_ubuntu_v01.08.03.89.AppImage %F
> Terminal=false
> PrefersNonDefaultGPU=true
> Icon=/home/path/BambuStudio.png
> Type=Application
> Hidden=false
> NoDisplay=false
> StartupNotify=false
> Terminal=false
> ```
> 
> Now Bambu Studio launches fast, no errors, and no annoying "X Stoped Responding" popup error at launch. Everything Renders as it should.
> 
> **But for Orca Slicer, I still get the "X Stoped Responding" popup error at launch. and when I do it on terminal I still see all the "Critical Error" freakout you see above that other have posted. The app renders now and it works fine. I don't even have webkit installed currently (I uninstalled for testing), but the "critial errors" are still there.**
> 
> Thank you for your assistance.
> 
> Will keep testing. Running Linux Mint 21.2 / NVIDIA GK107M [GeForce GTX 660M Mac Edition] Legacy Drivers 470 / iMac 27" (Late 2012) 2.9 GHz Core i5 (I5-3470S)

fixed issues on v2.0.0 also!

unhandled unexpected error on windows so built in vm, didnt run, app image had white display

JulioJair, 2024-04-22:
WEBKIT_DISABLE_COMPOSITING_MODE=1
works for me

closed 2024-07-29 by github-actions[bot]

trebory6, 2024-08-15:
Operating System: Nomara Linux 40
KDE Plasma Version: 6.1.3
KDE Development Platform: 4.14.38
Qt: 4.8.7
Kernel Version: 6.10.3-201.fsync.fc40.x86_64
Graphics Platform: Wayland

None of the solutions in this thread have worked for me. 

Is there a status on this issue?

KiwiHuman, 2024-12-09:
orca slicer starts showing stuff after these fixes for me, but now crashes after I click anything.