OrcaSlicer/OrcaSlicer #2705: Switching Machine Preset Changes Filament and Process Presets

CCS86, 2023-11-12:
### OrcaSlicer Version

1.8.0-rc2

### OS version

Win 10 x64

### Printer

X1C

### How to reproduce

1. Make and save a new user machine preset 
2. Start a new project
3. Create a new user process preset
4. Get multiple color user presets selected
5. Switch the machine preset

### Actual results

When you select a different machine preset, the process preset and the filament presets will change, erasing any part level multi-color selections you had made:

[image]

### Expected results

I expect to be able to change the machine preset without affecting the process or filament presets.

### Project file & Debug log uploads

..

### Checklist of files to include

- [ ] Log file
- [ ] Project file

CCS86, 2023-11-17:
Tested in Bambu Studio and cannot replicate. Seems isolated to OS.

poolboy97, 2023-11-20:
Having same issue. Came here to report same thing.

CCS86, 2023-12-05:
@SoftFever can you reproduce?

CCS86, 2023-12-14:
@SoftFever 

This bug is a major pain. Any chance you can look into a solution?

CCS86, 2024-01-07:
Still seems to be an issue with 1.9.0 final.

CCS86, 2024-01-21:
Man, over two months now and not even an acknowledgement. This bug is a real in the ass.

CCS86, 2024-02-20:
Still an issue with 1.9.1

@SoftFever any chance you can acknowledge this bug?

SoftFever, 2024-02-20:
Hi @CCS86

Orca remember filament and process configurations for each machine.
When you switch to a new machine, you can config them once for the new machine.
They will be remembered, and automatically changed when you switch between different machines in the future.

CCS86, 2024-02-20:
For me, this is not a great feature. I use different machine configurations depending on whether I want a longer dwell or no dwell for the bed to heat soak for a print. It's super disruptive to have it silently change my print and filament settings. It can either add or remove multiple filaments too.

Can we at least get a dialog to accept to decline that change?

CCS86, 2024-04-06:
@SoftFever 

This is still a real pain point! Here is another example:

You download a project file from Maker World, and open in Orca Slicer.  It is a multicolor print, with all the color changes configured. You start making some tweaks to the print settings and then save that as a "preset inside project". Then, you switch from the generic X1C printer profile, to your customized one...

Boom, everything is gone. All the filaments and color changes disappear. The preset inside project you saved disappears. You can't even undo in this case. Everything is gone.

This is not a good feature.

SoftFever, 2024-04-08:
@CCS86 
I've added an option to disable the per machine configuration behavior.
https://github.com/SoftFever/OrcaSlicer/pull/4930

Let me know if it works

CCS86, 2024-04-08:
> @CCS86 I've added an option to disable the per machine configuration behavior. #4930
> 
> Let me know if it works

Thank you!

closed 2024-04-09 by SoftFever

CCS86, 2024-04-20:
@SoftFever Testing the 2.1.0 nightly now.

While it seems to preserve the multi-color settings, it is still changing the print setting preset:

https://github.com/SoftFever/OrcaSlicer/assets/12439543/8e439a02-d45a-48fb-b122-0d0485b27f22

CCS86, 2024-04-20:
Even if I save the preset (in project, or global), then switch machines, I can't use that newly saved preset. It disappears I guess because it's not "compatible" with the new machine.

Maybe this is a separate issue, but I can't see why we should be locked out of selecting presets based on machine, especially when we are just talking about P1P, X1C, modified X1C.

CCS86, 2024-05-09:
@SoftFever , can we re-open this?

jwidess, 2024-06-13:
I'm also having the same problem, no dialog warning or anything when I switch printers. It seems to be the same problem as this person is having, https://www.reddit.com/r/BambuLab/comments/1afbsy8/retain_print_settings_when_switching_printers_in/

PR #4930 (2024-04-09): Add an option to allow user disable printer configuration
# Description
Add an option to allow user disable printer configuration
Fixes #4750
Fixes #2705
# Screenshots/Recordings/Graphs
[image]

## Tests

Tested with some multi color .3mf files