OrcaSlicer/OrcaSlicer #2707: v1.8.0-rc2 marked as "latest version" in github

core-code, 2023-11-12:
### OrcaSlicer Version

v1.8.0-rc2

### OS version

any

### Printer

none

### How to reproduce

go to https://github.com/SoftFever/OrcaSlicer/r

### Actual results

v1.8.0-rc2 marked as "latest version" in github

### Expected results

i expect all unstable (alpha/beta/rc) versions to be marked as "pre-release" in GitHub and only "stable" versions being marked as "Latest release"

### Project file & Debug log uploads

none

### Checklist of files to include

- [ ] Log file
- [ ] Project file

SoftFever, 2023-11-16:
I admit marking them as the latest release isn't the best practice. However, I'm trying to figure out a better way to expose these beta/RC versions to a wider user group. This way, more users might become aware of the latest updates and we could attract more volunteers willing to help with testing.

core-code, 2023-11-16:
thanks there. the issue here is also with downstream users (e.g.  Homebrew Cask, MacUpdater) that depend on the Github Atom Release feed only containing stable releases. i hope some other way to expose RC builds to widespread testing (by those that want to help testing) can be found ;-)

SoftFever, 2023-11-16:
That's a valid point.
Let's try not marking them as latest in the future