OrcaSlicer/OrcaSlicer #2708: Multiple height range modifiers not possible.

sylvled, 2023-11-12:
Hello !

**Is your feature request related to a problem? Please describe.**
I'm always frustrated when I need to add more than one height range modifier as I find nowhere any "Plus" sign. Right-clicking object and, in the popup menu, clickign "height range modifier" again is not adding another one. On the PrusaSlicer there is the "Plus Sign" that is working as I would like.

**Which printers will be beneficial to this feature**
All printers

**Describe the solution you'd like**
See first lines.

**Describe alternatives you've considered**
I did not find any alternative until now. Looking actively for it but still not found.

**Additional context**
French version of the 1.8.0-rc2 slicer (was same on older versions)

Thanks to the team !! This OrcaSlicer is a real awesome software !

Dominik1980, 2023-11-14:
[image]
Which language have you set? Maybe try changing the language. No “+” is displayed when I set the language to German. But it is shown in English version.

sylvled, 2023-11-15:
Great, I found it, now thanks to you @Dominik1980. Though this is still an issue as the + sign is not complete and is cut by the end of the left panel.
[image]
Maybe this is related to the screen resolution which is 4K ? I then changed to fullhd res, but no change. The panel seems to be unchangeable.
Unless I'm mistaking, this is now more a request to have a new feature (changing size of the left panel) than an issue now.
I will close this in a couple of days if there is no more comments.

Noisyfox, 2023-11-15:
I'll take a look.

closed 2023-11-16 by SoftFever

sylvled, 2023-11-17:
amazing is your reactivity ! Thanks and congrats !

ArahnDoberman, 2023-12-21:
I seem to run into the same problem with a 4k monitor at 150% UI Scale
Is there a potential workaround like a keyboard shortcut?
I tried Copy and paste which didn't work
[image]

Thank you :)

Noisyfox, 2023-12-21:
> I seem to run into the same problem with a 4k monitor at 150% UI Scale (running win 11 and OC 1.8.1) Is there a potential workaround like a keyboard shortcut? I tried Copy and paste which didn't work <img alt="Screenshot 2023-12-21 102141" width="338" src="https://private-user-images.githubusercontent.com/154508923/292157113-b8c91340-bd9a-4e5f-ba25-8a33df89a519.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTEiLCJleHAiOjE3MDMxNTc5MzksIm5iZiI6MTcwMzE1NzYzOSwicGF0aCI6Ii8xNTQ1MDg5MjMvMjkyMTU3MTEzLWI4YzkxMzQwLWJkOWEtNGU1Zi1iYTI1LThhMzNkZjg5YTUxOS5wbmc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBSVdOSllBWDRDU1ZFSDUzQSUyRjIwMjMxMjIxJTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDIzMTIyMVQxMTIwMzlaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT1lMjI5NDE3ZWRkZjE1N2VmZTAwZmEwMjFjYTlhNzAxYjVhNDU4YmU4ZDJkZTM4NTdkZTUzYTBkZDBiY2RkZWRhJlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCZhY3Rvcl9pZD0wJmtleV9pZD0wJnJlcG9faWQ9MCJ9.DlWHytr_qs_ceCos_4uF-ZgVgzR70-gS92w5Cyim1Oc">
> 
> Thank you :)

Select the Layer line instead.

yonkiman, 2025-09-27:
I just spent an embarrassing amount of time trying to add a second height modifier.  The - and + boxes looked like a way to adjust the height range of the first modifier, so it took me forever to realize that the + box was how I add a second modifier.

Would it be possible to have "Add <another> Height Range Modifier" an option when one right-clicks on the "Layers" line?  I think what confused me is that the way one adds a second (or more) height range modifier is _very_ different from the way one adds the first one.  

Cheers

PR #2756 (2023-11-16): Fix layer height modifier layout (#2708)
This fixes #2708

[image]