OrcaSlicer/OrcaSlicer #2709: K1 Max Camera view streched

niedzial1983, 2023-11-12:
### OrcaSlicer Version

1.8.0-rc2

### OS version

Window 11

### Printer

Creality K1 Max

### How to reproduce

When using DEVICE tab the camera view is streched out. 
[image]

### Actual results

The camera picture is streched out and not in correct resolution

### Expected results

Camrea window should be in correct resolution and size ( 1920*1080 )
[image]

### Project file & Debug log uploads

Attched is a picture from Orca ( streched ) and Creality Print  ( correct ) 

### Checklist of files to include

- [ ] Log file
- [ ] Project file

niedzial1983, 2023-11-12:
Attched is a picture from Orca ( s
[image]
treched ) and Creality Print ( correct )

closed 2024-02-20 by github-actions[bot]

X-Ryl669, 2024-11-04:
This is still not fixed. Please reopen.