OrcaSlicer/OrcaSlicer #2712: first layer z height issue and under extrusion on latest nightly 1.8.0 rc2

davidja92, 2023-11-13:
### OrcaSlicer Version

1.8.0

### OS version

windows 11

### Additional system information

11th gen 19-12900k 64gb ram nvidia rtx 3080

### Printer

bambu X1C

### How to reproduce

print issues occur on latest nightly but go away on a 1.7.0 nightly with same settings same print same everything. i have both gcodes attached one is on 1.7.0 the other is on 1.8.0 latest nightly build orca #650 nightly 

### Actual results

first layer is bad like nozzle too far from bed and there is under extrusion higher up on the print.

### Expected results

perfect bed level perfect print

### Project file & Debug log uploads

[image]
[image]
[orca 1.7.0 and 1.8.0 abs.zip](https://github.com/SoftFever/OrcaSlicer/files/13330133/orca.1.7.0.and.1.8.0.abs.zip)

### Checklist of files to include

- [ ] Log file
- [X] Project file

R4toLoko, 2023-11-13:
I am having similar problem, and no matter what I check on my printer, I cannot find a single screw out of place.
It could be we are on the same boat or it could be I do have a different problem.

1.8.0-beta nightly, notice how the right model looks like crappy with gaps everywhere on the first layer while the left one is that I am used to.

[image]

I print this 200x200x0.2 @ 0.2 layer height to check my first layer.
It just gets worse and worse over and over like if my Z-offset is all over the place.
But it makes no sense, I have a Voron Trident, z-offset issues is something it doesn't know the meaning of it and I have been dealing with this problem for weeks now.

But again, it could be just coincidence and I have a bigger problem.
I will do some tests tonight.

[image]

davidja92, 2023-11-13:
> I am having similar problem, and no matter what I check on my printer, I cannot find a single screw out of place. It could be we are on the same boat or it could be I do have a different problem. [...]

I see the layer squish issue not as bad as mine but I see it

R4toLoko, 2023-11-13:
> I see the layer squish issue not as bad as mine but I see it

That is the best one, you should see others.
I have done first layer calibration over and over, I used 2x half spool printing this cube over and over, I disassembled my entire hotend looking for something off, you name it. 
Only a cube showed a weird sign of under extrusion, my main problem is the z-offset. it is basically all over the place.

I am so hoping this is the root cause, I would be less pissed lol

davidja92, 2023-11-13:
newest nightly under 1.8.0 rc2 https://github.com/SoftFever/OrcaSlicer/actions/runs/6841588077 works great. this commt https://github.com/SoftFever/OrcaSlicer/commit/15358b211d5c352fa6d72eff00859629e648b750 fixes the issue i can consider this closed

closed 2023-11-13 by davidja92