OrcaSlicer/OrcaSlicer #2713: Build plate appears black in Creality K1.

wichos12, 2023-11-13:
### OrcaSlicer Version

1.8.0 - rc

### OS version

windows 11 Home 23H2

### Additional system information

CPU: AMD Ryzen 5 5500U with Radeon Graphics
GPU: AMD Radeon(TM) Graphics
Memory: 128 GB ROM / 20 GB RAM

### Printer

Creality K1

### How to reproduce

1. Select the Creality K1 printer.
2. Go to the prepare window.

### Actual results

The build plate appears with the correct texture and model for a few seconds, then it turns black and stays that way. It also appears black in the preview window.
This also happens with the Elegoo Neptune 4 Pro printer. So far, I have only identified the issue in these two printers.
[image]

### Expected results

The build plate stay with the rigth texture and model
*Image only for reference
[image]

### Project file & Debug log uploads

[log.zip](https://github.com/SoftFever/OrcaSlicer/files/13330440/log.zip)
[Proyect.zip](https://github.com/SoftFever/OrcaSlicer/files/13330441/Proyect.zip)

### Checklist of files to include

- [X] Log file
- [X] Project file

just-trey, 2023-11-16:
@wichos12  I stumbled across this, and I don't have this issue. 1.8.0 - rc2 is out. I wonder if that solves the issue. It could be an and thing too. That is the only difference I see in our systems. 
[image]

Ocraftyone, 2023-11-16:
It does not solve the issue. AFAIK, this is an issue with AMD graphics cards. It seems to be a transparency issue with png images used as the build plate image. I have not gotten around to troubleshooting via lookING through the source code, but I have observed that it seems svg formated images seem to work correctly.

Noisyfox, 2023-11-17:
Could you try the dev build from https://github.com/SoftFever/OrcaSlicer/pull/2603 and see if it has the same issue? Thanks.

Genuwine1976, 2023-11-17:
I'm having the same issue. Was a fix for this issues found, and if so how do I fix it?

Ocraftyone, 2023-11-18:
@Noisyfox I too had tried that PR and had no change.

Noisyfox, 2023-11-18:
Cool, I'll take a look then.

Noisyfox, 2023-11-18:
Once the build of PR https://github.com/SoftFever/OrcaSlicer/pull/2780 has done could you give it a try and see if that helps? I currently don't have my AMD laptop around so couldn't test for myself.

PR #2780 (2023-11-19): Fix PNG build plate texture not rendering on AMD GPUs
This should fix #2713
Also fix #2782

cherry picked from commit prusa3d/PrusaSlicer@971f2a08e2758ea1d91e75af9b1443d14b41f895
cherry picked from commit prusa3d/PrusaSlicer@eee4453993ba38f3f11c8f89c56cefa50eaaf74b
closed 2023-11-19 by SoftFever