OrcaSlicer/OrcaSlicer #2714: CLI OrcaSlicer crashes to desktop on Fedora

eropple, 2023-11-13:
### OrcaSlicer Version

1.7.0 and 1.8.0-rc2

### OS version

Fedora 39

### Additional system information

CPU: Ryzen 3900X
RAM: 64GB DDR4-3000
GPU: nVidia 2070 Super

### Printer

n/a

### How to reproduce

Attempt to run a command-line operation on a mesh with OrcaSlicer. `--export-stl` and `--export-3mf` do not change the behavior here (it's pretty clearly failing earlier in the process). Even a no-op method that should effectively just load and save the mesh fails.

I can't get a local build of OrcaSlicer to actually work, or I'd try to debug further. AppImages aren't a friendly way to debug.

### Actual results

For **v1.7.0**:

```
$ ~/Downloads/OrcaSlicer_Linux_V1.7.0.AppImage --export-stl  --outputdir ./output input/foo.stl 
[2023-11-12 23:38:41.494086] [0x00007f87d9d17ec0] [trace]   Initializing StaticPrintConfigs
[2023-11-12 23:38:41.607668] [0x00007f87d9d17ec0] [error]   calc_exclude_triangles:Unable to create exclude triangles

[1]    129887 segmentation fault (core dumped)  ~/Downloads/OrcaSlicer_Linux_V1.7.0.AppImage --export-stl --outputdir ./outpu
```

For **v1.8.0-rc2**:

```
$ ~/Downloads/OrcaSlicer_Linux_V1.8.0-rc2.AppImage --export-stl  --outputdir ./output input/foo.stl      
[2023-11-12 23:37:53.419399] [0x00007f87f81fdec0] [trace]   Initializing StaticPrintConfigs
[2023-11-12 23:37:53.545204] [0x00007f87f81fdec0] [error]   calc_exclude_triangles:Unable to create exclude triangles

(process:129831): Gdk-CRITICAL **: 23:37:53.552: gdk_screen_get_root_window: assertion 'GDK_IS_SCREEN (screen)' failed

(process:129831): Gdk-CRITICAL **: 23:37:53.552: gdk_cairo_create: assertion 'GDK_IS_WINDOW (window)' failed

(process:129831): Gdk-CRITICAL **: 23:37:53.552: gdk_window_get_width: assertion 'GDK_IS_WINDOW (window)' failed

(process:129831): Gdk-CRITICAL **: 23:37:53.552: gdk_window_get_height: assertion 'GDK_IS_WINDOW (window)' failed

(process:129831): GLib-GObject-CRITICAL **: 23:37:53.554: invalid (NULL) pointer instance

(process:129831): GLib-GObject-CRITICAL **: 23:37:53.554: g_signal_connect_data: assertion 'G_TYPE_CHECK_INSTANCE (instance)' failed

(process:129831): Gtk-CRITICAL **: 23:37:53.554: _gtk_css_lookup_resolve: assertion '(((__extension__ ({ GTypeInstance *__inst = (GTypeInstance*) ((provider)); GType __t = ((_gtk_style_provider_private_get_type ())); gboolean __r; if (!__inst) __r = (0); else if (__inst->g_class && __inst->g_class->g_type == __t) __r = (!(0)); else __r = g_type_check_instance_is_a (__inst, __t); __r; }))))' failed

(process:129831): GLib-GObject-CRITICAL **: 23:37:53.554: g_object_set_data_full: assertion 'G_IS_OBJECT (object)' failed

(process:129831): Gtk-ERROR **: 23:37:53.554: Can't create a GtkStyleContext without a display connection
[1]    129831 trace trap (core dumped)  ~/Downloads/OrcaSlicer_Linux_V1.8.0-rc2.AppImage --export-stl --outputdir 
```

### Expected results

Not to have a CLI application crash out because of GTK3 errors.

Bambu Studio v01.08.00.57 does not exhibit similar behavior.

### Project file & Debug log uploads

No debug logs are generated.

### Checklist of files to include

- [ ] Log file
- [ ] Project file

major-tweaks, 2023-11-26:
I can fully support that. Same crash on (X)ubuntu 20.04.6 LTS

./OrcaSlicer_Linux_V1.8.0.AppImage 
[2023-11-26 17:24:13.574639] [0x00007f5f60a3edc0] [trace]   Initializing StaticPrintConfigs
add font of HarmonyOS_Sans_SC_Bold returns 1
add font of HarmonyOS_Sans_SC_Regular returns 1
add font of NanumGothic-Regular returns 1
add font of NanumGothic-Bold returns 1

(orca-slicer:4667): Gtk-CRITICAL **: 17:24:14.007: gtk_box_gadget_distribute: assertion 'size >= 0' failed in GtkScrollbar
loading network module at /home/xyz/.config/OrcaSlicer/plugins/libbambu_networking.so
error, dlerror is /home/xyz/.config/OrcaSlicer/plugins/libbambu_networking.so: Kann die Shared-Object-Datei nicht öffnen: Datei oder Verzeichnis nicht gefunden
after dlopen, network_module is (nil)

(orca-slicer:4667): Gtk-CRITICAL **: 17:24:15.555: gtk_window_resize: assertion 'width > 0' failed

(orca-slicer:4667): GLib-GObject-WARNING **: 17:24:15.577: invalid cast from 'wxPizza' to 'GtkCellLayout'

(orca-slicer:4667): Gtk-CRITICAL **: 17:24:15.577: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed

(orca-slicer:4667): GLib-GObject-WARNING **: 17:24:15.578: invalid cast from 'wxPizza' to 'GtkCellLayout'

(orca-slicer:4667): Gtk-CRITICAL **: 17:24:15.578: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed

(orca-slicer:4667): GLib-GObject-WARNING **: 17:24:15.580: invalid cast from 'wxPizza' to 'GtkCellLayout'

(orca-slicer:4667): Gtk-CRITICAL **: 17:24:15.580: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed

(orca-slicer:4667): GLib-GObject-WARNING **: 17:24:15.581: invalid cast from 'wxPizza' to 'GtkCellLayout'

(orca-slicer:4667): Gtk-CRITICAL **: 17:24:15.581: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed

(orca-slicer:4667): GLib-GObject-WARNING **: 17:24:15.583: invalid cast from 'wxPizza' to 'GtkCellLayout'

(orca-slicer:4667): Gtk-CRITICAL **: 17:24:15.583: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed

(orca-slicer:4667): GLib-GObject-WARNING **: 17:24:15.585: invalid cast from 'wxPizza' to 'GtkCellLayout'

(orca-slicer:4667): Gtk-CRITICAL **: 17:24:15.585: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed

(orca-slicer:4667): GLib-GObject-WARNING **: 17:24:15.588: invalid cast from 'wxPizza' to 'GtkCellLayout'

(orca-slicer:4667): Gtk-CRITICAL **: 17:24:15.588: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed

(orca-slicer:4667): GLib-GObject-WARNING **: 17:24:15.589: invalid cast from 'wxPizza' to 'GtkCellLayout'

(orca-slicer:4667): Gtk-CRITICAL **: 17:24:15.589: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed

(orca-slicer:4667): GLib-GObject-WARNING **: 17:24:15.591: invalid cast from 'wxPizza' to 'GtkCellLayout'

(orca-slicer:4667): Gtk-CRITICAL **: 17:24:15.591: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed

(orca-slicer:4667): GLib-GObject-WARNING **: 17:24:15.592: invalid cast from 'wxPizza' to 'GtkCellLayout'

(orca-slicer:4667): Gtk-CRITICAL **: 17:24:15.592: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed

(orca-slicer:4667): Gtk-CRITICAL **: 17:24:15.699: gtk_window_resize: assertion 'height > 0' failed

(orca-slicer:4667): Gtk-CRITICAL **: 17:24:15.706: gtk_widget_set_size_request: assertion 'width >= -1' failed

(orca-slicer:4667): Gtk-CRITICAL **: 17:24:15.706: gtk_widget_set_size_request: assertion 'width >= -1' failed

(orca-slicer:4667): Gtk-CRITICAL **: 17:24:15.706: gtk_widget_set_size_request: assertion 'width >= -1' failed

(orca-slicer:4667): Gtk-CRITICAL **: 17:24:15.706: gtk_widget_set_size_request: assertion 'width >= -1' failed

(orca-slicer:4667): Gtk-CRITICAL **: 17:24:15.706: gtk_widget_set_size_request: assertion 'width >= -1' failed

(orca-slicer:4667): Gtk-CRITICAL **: 17:24:15.706: gtk_widget_set_size_request: assertion 'width >= -1' failed

(orca-slicer:4667): Gtk-CRITICAL **: 17:24:15.706: gtk_widget_set_size_request: assertion 'width >= -1' failed

(orca-slicer:4667): Gtk-CRITICAL **: 17:24:15.706: gtk_widget_set_size_request: assertion 'width >= -1' failed

(orca-slicer:4667): Gtk-CRITICAL **: 17:24:15.717: gtk_window_resize: assertion 'height > 0' failed

(orca-slicer:4667): Gtk-CRITICAL **: 17:24:15.718: gtk_widget_set_size_request: assertion 'width >= -1' failed

(orca-slicer:4667): Gtk-CRITICAL **: 17:24:15.718: gtk_widget_set_size_request: assertion 'width >= -1' failed
Speicherzugriffsfehler (Speicherabzug geschrieben)

DiegoHerrera1890, 2024-02-14:
Hello Sir,
I am new using OrcaSlicer and I'd like to run it using CLI commands, do you know where I can learn or read the commands? or how to use CLI? Thank you so much.

Lord3n20, 2024-04-19:
> I can fully support that. Same crash on (X)ubuntu 20.04.6 LTS
>
> ./OrcaSlicer_Linux_V1.8.0.AppImage [2023-11-26 17:24:13.574639] [0x00007f5f60a3edc0] [trace] Initializing StaticPrintConfigs add font [...]

Same for me. Is it a user error or still on its way to get fixed?
I am running Arch with dwm thought

closed 2024-07-27 by github-actions[bot]

IllogicalDesigns, 2025-09-09:
Still an issue on arch linux

p0lp44r, 2026-01-15:
> Still an issue on arch linux

Same here on Arch. With AUR Package orca-slicer-bin-2.3.1-7