OrcaSlicer/OrcaSlicer #2719: Re-add instancing support for objects on the print bed

mlewis-everley, 2023-11-13:
**Is your feature request related to a problem? Please describe.**
Prusa Slicer allows adding of "instances" of an object (right click and object on the bed and select "add instance"). For some reason this no longer appears to be possible in Bambu/Orca slicer?

The is an issue logged for Bambu slicer here: https://github.com/bambulab/BambuStudio/issues/496 That seems to discuss the topic extensively. 

As mentioned in the topic, I can clone objects, cloned objects behave differently to instances. A basic example would be that if I clone an item with paint on supports, and I adjust the supports, I will need to re-adjust, or re-clone (and re-arrange) the relevant items.

Instances would allow me to just edit the parent item and all instances would then be affected.

**Which printers will be beneficial to this feature**
All?

**Describe the solution you'd like**
Re-add that ability to add instances of an object, as PrusaSlicer does.

erlis, 2023-11-13:
+1 to this. The add instance is really useful and I really miss this. My use case is that I have a small part that I need to print a lot of them and I need to "split into objects" oh boy, now I cannot just split the parent object, I need to split one by one and if I do a modifier like, block support then I can't split object... big mess, so far my only solution is to do the split and modifier to all objects one by one... big pain, and the only reason I still use Prusa Slicer, I'm not going to do this manually to 20 objects on the bed....

pennystockss, 2023-11-14:
It's so painful. With several plates full of small objects Bambu/Orca could want to use 40+gb of ram and is so sluggish, even with the available memory. Not to mention the file sizes are insane.

paulorlyk, 2023-11-26:
It's just inconvenient to print many instances of the same object: There is no easy control for the number of instances - instead of just pressing + or - you need to count them manually and add or remove clones trough context menus. And, most of all, if any feature of the original object changes, custom supports for example, it does not propagate to the clones, and you need to repopulate your bed manually all over again.
Please add the instancing support back - it makes life much simpler when mass-producing parts.

henleyb, 2024-03-02:
I agree, it's a huge pain to paint an object and then spend the time cloning and arranging carefully on a plate only to slightly update the design and instead of just replacing one and all its instances also updating I have to basically start the whole process from scratch, it's a huge time waster for me and I dread making any changes.

mlewis-everley, 2024-06-02:
I am just going to bump this as it has gone stale, would be great to hear from one of the core devs regarding feasibility?

CCS86, 2024-06-07:
@SoftFever any interest in implementing this?

It's a massive pain point when working with duplicate objects. Having to either manually paint every seam / support, manually reload every single object when the geometry model gets updated, etc.

morganvenable, 2024-06-28:
Hugely valuable feature for production.  🙏🏻🙏🏻🙏🏻

morganvenable, 2024-06-28:
Honestly this is worth enough time to me and so many other people that I'd put up a cash bounty on it.  $1000 when I can use it -- cash in hand, by whatever means necessary.
Anyone else willing to chip in to emphasize the value?

SupItsSaige, 2024-08-17:
OMG! Looks like it should get added soon at least based on the pull request! THANK GOD! I am trying to switch from PS to Orca (but waiting until Prusa XL 5T is supported). If there were no instances, I would have 100% stayed with PS. That functionality is so important when you are printing a lot of objects and making lots of changes. Glad to see it should come to back to Orca. Exciting!

morganvenable, 2024-08-17:
❤️ ❤️ ❤️

morganvenable, 2024-08-25:
I'm using this right now and so happy to have it back.  However, the "+" button also invokes auto-arrange, which is very slow for large numbers of the same part.  This wasn't the case from what I recall in Prusaslicer before -- is this something that can be disabled so the user can call autoarrange only when needed, or use the same placement logic as copy/paste?

CCS86, 2024-08-28:
Sweet!

erlis, 2024-10-10:
@morganvenable  I'm curios to know what version are you using, I just installed the 2.2.0-beta and this is still missing.

morganvenable, 2024-10-10:
You have to use a dev build to test it.  Unfortunately there's still a ways
to go -- the main use of instances for me is keeping physical layouts of
parts on plates while updating all instances of a given part, and the
progress so far doesn't quite preserve that stuff.  So for big arrays of
carefully arranged things, it's still not super usable.

Some of the arrangement preservation/creation behaviors for copy/paste
clones (vs instances) also don't work as expected yet.  Hopefully it'll get
there eventually.

kozakm00, 2025-01-22:
Yes, please. Just copy it from Prusa slicer

Tupson444, 2025-03-19:
I can't believe this function is missing. I'm using PrusaSlicer, but I like more options available in Orca so I'm trying it out. However, Orca is crippled by lacking this essential, basic feature... so never mind, I'll keep using PS for most of my prints.

Tupson444, 2025-06-18:
Still wanted. I'm literally not using Orca because of this (I tried it, then went back to PrusaSlicer when I saw it doesn't have this function).

HB-Stratos, 2025-08-09:
I would also like to see this re-implemented. I used it a lot on PrusaSlicer before I transitioned.

positron96, 2025-08-31:
well, lack of this is a deal-breaker for me

closed 2025-10-26 by SoftFever