OrcaSlicer/OrcaSlicer #2720: Filename Placeholder  "initial_filament_type" not available

Sidirius, 2023-11-13:
### OrcaSlicer Version

1.8.0-rc2

### OS version

macOS 14.0

### Additional system information

Apple Silicon M1 Pro

### Printer

Voron 2.4

### How to reproduce

Add the variable [initial_filament_type] or {initial_filament_type} to the "Filename format", slice a file, press print oder export gcode.

### Actual results

Error Message: Variable does not exist

### Expected results

Add the initial_filament_type to the filename.

### Project file & Debug log uploads

[OrcaCube_BugReport.3mf.zip](https://github.com/SoftFever/OrcaSlicer/files/13334274/OrcaCube_BugReport.3mf.zip)

[Archive.zip](https://github.com/SoftFever/OrcaSlicer/files/13334293/Archive.zip)

### Checklist of files to include

- [X] Log file
- [X] Project file

Sidirius, 2023-11-13:
Filename placeholder list by PrusaSlicer: [https://help.prusa3d.com/de/article/liste-der-platzhalter-variablen_205643](https://help.prusa3d.com/de/article/liste-der-platzhalter-variablen_205643)
Placeholder list by Bambu Studio: [https://wiki.bambulab.com/en/software/bambu-studio/placeholder-list](https://wiki.bambulab.com/en/software/bambu-studio/placeholder-list)

kieraneglin, 2024-01-04:
I got around this with `filament_type[initial_tool]`

Sidirius, 2024-01-11:
> {filament_type[initial_tool]}

For the GCODE there is no prob, works like expected. What I mean is for the GCODE-Filename. I would like to have my initial filament (and tool, but that works) in the filename.

kieraneglin, 2024-01-11:
@Sidirius this also works in the filename - I'm using it!

Sidirius, 2024-01-11:
> @Sidirius this also works in the filename - I'm using it!

Can you send me your filename format, I tested it with the stub ({filament_type[initial_tool]}) and got an error when I tried to export…

kieraneglin, 2024-01-11:
@Sidirius `{input_filename_base}_{nozzle_diameter[0]}_{filament_type[initial_tool]}.gcode` and I'm on Orca 1.8.1
[image]

Sidirius, 2024-01-12:
> @Sidirius `{input_filename_base}_{nozzle_diameter[0]}_{filament_type[initial_tool]}.gcode` I've tested on Orca 1.8.1 and 1.9.0 and both have worked (but this has been working for me for many months) ![Screenshot 2024-01-11 at 10 55 09 AM](https://private-user-images.githubusercontent.com/569917/296021317-f7646c8d-01ed-4bb0-a0ce-ceea98eca3ca.jpg?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MDUwNDM3NDYsIm5iZiI6MTcwNTA0MzQ0NiwicGF0aCI6Ii81Njk5MTcvMjk2MDIxMzE3LWY3NjQ2YzhkLTAxZWQtNGJiMC1hMGNlLWNlZWE5OGVjYTNjYS5qcGc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQwMTEyJTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MDExMlQwNzEwNDZaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT03MTM1Nzk2N2ExZjkxZWE2ODY0YWMyZjYxNTQ0NDE2MDQ2MzUyYWQ1NWFhMzZkZmQyODQ5NDdhZmNkOGVhNzMyJlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCZhY3Rvcl9pZD0wJmtleV9pZD0wJnJlcG9faWQ9MCJ9.wTy1ByTOE-Z61p-px7KpjHNQWAJ6fBieYaBjz50FFyA)

Thanks mate, now it works, now it also works when I change it in my stub.... I have no clue what went wrong yesterday...

closed 2024-04-19 by github-actions[bot]