OrcaSlicer/OrcaSlicer #2723: Property invisible - Don't support bridges

Shama71, 2023-11-13:
### OrcaSlicer Version

1.8.0.-rc2

### OS version

Windows 11

### Printer

Any

### How to reproduce

Click button "Search in preset"
Type "bridge".
Select "Advanced: Don't support bridges"
[image]

### Actual results

[image]

### Expected results

Expected property.

### Project file & Debug log uploads

[Untitled.zip](https://github.com/SoftFever/OrcaSlicer/files/13335138/Untitled.zip)

### Checklist of files to include

- [ ] Log file
- [ ] Project file

Noisyfox, 2023-11-13:
This option is not available for certain types of support, so it's hidden. Currently it creates some confusion and #2170 means for solving this.

Shama71, 2023-11-13:
Thanks for answer.

"Max bridge length" not working.
Remove Support blocker.
[Fan.zip](https://github.com/SoftFever/OrcaSlicer/files/13335551/Fan.zip)

Noisyfox, 2023-11-13:
That only works well for large wide bridges, which will prevent support at the center of the bridge. In your case it's actually trying to support the loops around the bridge, which are categorized as overhang walls not bridges.

Shama71, 2023-11-14:
Thanks for answer.

closed 2023-11-14 by Shama71