OrcaSlicer/OrcaSlicer #2724: Wiki links in software don't function as expected

jwvanderbeck, 2023-11-13:
### OrcaSlicer Version

1.8.0-beta2

### OS version

Windows 11

### Additional system information

CPU: AMD Ryzen 7950X3D
GPU: NVidia RTX 4090
RAM: 64gb

### Printer

Bambu Labs X1C

### How to reproduce

1. Go to print settings
2. Find a setting that is underlined indicating it should open an information page on the wiki when clicked
3. Click the setting

### Actual results

The wiki will open at the proper link, but then immediately redirect back to the front page of the wiki with a message in the top indicating the user does not have permission to update the wiki.

[image]

### Expected results

The proper wiki page with information on the setting should open.

### Project file & Debug log uploads

[Spool_Weight - Original_debug.zip](https://github.com/SoftFever/OrcaSlicer/files/13335196/Spool_Weight.-.Original_debug.zip)
[log.zip](https://github.com/SoftFever/OrcaSlicer/files/13335204/log.zip)

### Checklist of files to include

- [X] Log file
- [X] Project file

Bronsens, 2023-12-01:
I have the same problem but it is the entire wiki.

LupusTheCanine, 2023-12-16:
I have the same problem.

mseymour, 2024-01-10:
This is still an issue, I get "You do not have permission to update this wiki." when trying to access any of the pages linked through the slicer settings.

gshpychka, 2024-01-24:
@SoftFever please take a look at this, all wiki links are broken.

PeterKjaerFisker, 2024-03-06:
I have the problem as well, no links from the Orca Slicer software works for me

gshpychka, 2024-06-05:
This is still relevant

SoftFever, 2024-06-05:
Hey guys,

That's not a bug.
Not all parameters are documented at the moment.
That require tremendous (collective) effort

SoftFever, 2024-06-06:
There's no upstream WIKI repo unfortunately.

jwvanderbeck, 2024-06-06:
The original ticket was not about missing wiki entries, it was about the fact that the software did not correctly open the already existing links.  The entries DID exist.

I believe that has been fixed however.