OrcaSlicer/OrcaSlicer #2726: add shrinkage calibration to the available calibration functions

julie777, 2023-11-13:
## Is your feature request related to a problem? Please describe.
In many cases a little shrinkage of the print does not matter. However, in other applications, such as structural parts, the user will want the printed part to match the model dimensions as closely as possible. As a novice 3D printer user I would often change the model dimensions to get the desired output. Now I manually adjust the shrinkage for each filament type.

Granted, the shrinkage might be quite small. For example with Hatchbox PLA in my current model 200mm prints as 199.56mm. However inaccuracies can accumulate with multiple parts and then the final result does not fit correctly.

## Which printers will be beneficial to this feature
All

## Describe the solution you'd like
I use a simple printed rectangle to measure and set shrinkage. It does not take much time or filament.
[image]
It is 200mm square (for P1P) and the sides are 5mm x 5mm.

It would be useful if the user could specify the x and y dimensions, which allows use on any printer.

## Describe alternatives you've considered
As above I just do it manually.

## Additional context

In addition, as I have recommended before for all calibration functions, it would be nice if the calibration function would 
1. provide a dialog for parameters
2. provide a dialog after printing describing how the user should interpret results and allow entering measurements or other information
3. perform any required calculations to determine the best filament settings
4. and update the filament profile directly
5. show the filament profile with the changes to allow the user to confirm/save the changes

jrm93, 2023-11-13:
Very nice idea but a 200x200 mm frame is a bad idea imo because most people have 150mm calipers. Something like [this ](https://www.printables.com/model/556196-calibrating-core-xy-printers-for-dimensional-accur)or [this ](https://www.printables.com/model/239603-100x100x100-calibration)will be more suitable for this kind of calibration.

julie777, 2023-11-13:
I was just thinking more about this. I believe that this should be the last calibration performed. Other calibrations will certainly have an affect on dimensions.

In addition, the corners of the test print should have a fillet added to prevent any excessive pressure advance from affecting measurements when calipers are used across the corners.

julie777, 2023-11-13:
> Very nice idea but a 200x200 mm frame is a bad idea imo because most people have 150mm calipers. Something like [this [...]

I did mention that the user should be able to set the dimensions. This is important both due to printer size and the ability to accurately measure the printed part. You are correct that most users are more likely to have 6"/150mm calipers. I just happened to be writing this after measuring an actual part that was 200mm. I do believe that the larger the test part is, the more accurate the compensation will be.

SoftFever, 2023-11-14:
Maybe a 120x120mm steel square shape would be perfect?

asychev, 2023-11-14:
IMO 100x30x3 with round corners is more than enough and fits almost all printers, including small ones like V0.
100 on the long side is also easy for calculations.

Eldenroot, 2023-12-02:
120x120mm would be nice

LupusTheCanine, 2024-01-02:
I would like to propose producing stepped calibration ladder.
Something like:
[image]
These are 50 and 200 mm ladders.
IMHO it is best to print the ladder at +1 step than your intended measurements because then it is easier to measure the last step.

Here is example SCAD code to generate ladders. I personally prefer to print two at a time for a given size.

```
size=100;
step=10;
height=0.8;

// size is a vector [w, h, d]
module roundedBox(size, radius)
{
	rot = [ [0,0,0], [90,0,90], [90,90,0] ];
    cube(size - [2*radius,0,0], true);
    cube(size - [0,2*radius,0], true);
    for (x = [radius-size[0]/2, -radius+size[0]/2],
         y = [radius-size[1]/2, -radius+size[1]/2]) {
        translate([x,y,0]) cylinder(r=radius, h=size[2], center=true);
    }
}

$fn=40;

color("Yellow");
stepCount=ceil(size/step);
for(i=[0:stepCount]){
    translate([0,0,i*height+height])
    roundedBox([5,step*(stepCount-i+1),height], 1.5);
}
```

And here is example deviation result for my printer
[image]

I need to do the same test for the gaps though.

closed 2024-04-09 by github-actions[bot]