OrcaSlicer/OrcaSlicer #2730: MMU Print Issue

TheAtomicSoul, 2023-11-14:
### OrcaSlicer Version

1.8.0, 1.8.0r2

### OS version

Windows11

### Printer

Bambu p1p

### How to reproduce

Solid MMU model is having issues where two colors print next to each other.  This is happening on both 1.8.0 and 1.8.0r2. 
v1.7.0 does not have this issue.

1st layer
[image]
[image]

2nd layer
[image]
[image]

### Actual results

edges at color borders inside the model are chaotic. 

### Expected results

smoother edges/borders like v1.7.0

### Project file & Debug log uploads

na

### Checklist of files to include

- [X] Log file
- [ ] Project file

SoftFever, 2023-11-14:
can you share a sample project?

TheAtomicSoul, 2023-11-15:
> can you share a sample project?

[image]

TheAtomicSoul, 2023-11-15:
[error Cube.zip](https://github.com/SoftFever/OrcaSlicer/files/13359398/error.Cube.zip)

TheAtomicSoul, 2023-11-15:
> > can you share a sample project?
> 
> [image]

Sorry for the close/open, did in error. 

On the cube, I did "Paint by height range", turned it sideways, and it was fine. Then I cleared the paint and did the above paint job manually, and got the error. 

Going back in, I can paint by height and turn it sideways and it's fine. Add in manual lines with the curcle tool and it goes AWOL. 
[image]

SoftFever, 2023-11-16:
I don't quite understand your question.

This is what I see from the attached project.
As far as I can see, it looks correct?
[image]

TheAtomicSoul, 2023-11-17:
> I don't quite understand your question.
> This is what I see from the attached project. As far as I can see, it looks correct?

Can you show what the bottom layer looks like? 

Even on mine the view of the side wall looks right, but the bottom borders are extremely jagged. 

The paintbursh paints with "squared" edges, rather than a smooth one, when you use a larger brush. I think that's what causing the edges I see on the bottom. (my guess)