OrcaSlicer/OrcaSlicer #2733: Cannot slice large object - which works fine in Bambu Studio

barsk, 2023-11-14:
### OrcaSlicer Version

1.8.0 - Beta2

### OS version

Windows 10

### Printer

Bambu Labs P1S

### How to reproduce

Trying to slice the Babo drawer module 30, plate 2 (the base frame), gives an error that the object is "too close to the exclusion area". There is no way to position the object with Orcaslacer that does not give an error since the size of the object is maximized to use the max area.

With Bambu Studio there is no error with the default placement from the .3mf project file. All printer settings in the Bambu Lab P1S "Basic Information" tab  is default and equal between Orcaslicer and Bambu Studio.

### Actual results

 An error that the object is "too close to the exclusion area".

### Expected results

No error is expected.

### Project file & Debug log uploads

[https://www.printables.com/model/576213-babo-drawer-modul-30](https://www.printables.com/model/576213-babo-drawer-modul-30)

[babo-drawer-modul-30-model_files.zip](https://github.com/SoftFever/OrcaSlicer/files/13347586/babo-drawer-modul-30-model_files.zip)

### Checklist of files to include

- [ ] Log file
- [X] Project file

Noisyfox, 2023-11-14:
You have to place the model very carefully into a specific location and it will allow you to print:
[image]
Make sure both left and right have enough space.

barsk, 2023-11-14:
It appears you are right! I did try to carefully move it, but did not find a position that worked. 
Funny though that the project file that positions the objects works fine on Bambu studio but not Orcaslicer. That is maybe not a bug but a nuisance.

closed 2023-11-14 by barsk