OrcaSlicer/OrcaSlicer #2739: Cosmetic toolbar appearance

66mkiqdv, 2023-11-14:
### OrcaSlicer Version

1.8.0-rc2

### OS version

openSUSE Tumbleweed

### Additional system information

## Hardware Information:
- **Hardware Model:**                       ASUS All Series
- **Memory:**                                      8.0 GiB
- **Processor:**                                   Intel® Core™ i5-4670K × 4
- **Graphics:**                                    AMD Radeon™ RX 570 Series
- **Disk Capacity:**                            1.9 TB

## Software Information:
- **Firmware Version:**                     1402
- **OS Name:**                                    openSUSE Tumbleweed
- **OS Build:**                                     (null)
- **OS Type:**                                      64-bit
- **GNOME Version:**                        45.1
- **Windowing System:**                   Wayland
- **Kernel Version:**                           Linux 6.5.9-1-default

### Printer

Creality Ender 3v2

### How to reproduce

New project > View > Use orthogonal view = ugly toolbar icons.

### Actual results

Orthogonal view messes with the toolbar icons, perspective view does not. Same issue is present in Bambu Studio.

Perspective view: 
[image]
Orthogonal view:
[image]

### Expected results

Toolbar icon appearance is expected to remain the same regardless of view setting.

### Project file & Debug log uploads

Files not necessary.

### Checklist of files to include

- [ ] Log file
- [ ] Project file

Noisyfox, 2023-11-15:
Could you give this build a try?
https://github.com/SoftFever/OrcaSlicer/pull/2603/checks

66mkiqdv, 2023-11-15:
Build tried and the cosmetic distortion now affects both orthogonal and perspective views.

[image]

Noisyfox, 2023-11-16:
I see. Will try fixing this.

Noisyfox, 2023-11-18:
Could you try this build? https://github.com/SoftFever/OrcaSlicer/actions/runs/6912441593

66mkiqdv, 2023-11-19:
I have tried the latest build you suggested and can confirm that the cosmetic distortion is no longer present, both perspective and orthogonal view look good. Nice job.

closed 2023-11-19 by 66mkiqdv