OrcaSlicer/OrcaSlicer #2744: "Raft" is not in the correct menu

DVSVIDEO, 2023-11-15:
"Raft" is not in the correct menu. "Raft" in other slicers is correctly in the "Bed adhesion" menu.
This is where it belongs, according to its role
Currently located under the "Support" tab in the "Bed adhesion" section.
[image]
**Describe the solution you'd like**
It should be moved to the "Bed adhesion" section under the "others" tab.
Instead of "Brim type", "Adhesion type" would be more appropriate and the "Raft" menu item should be placed in the drop down menu.
Raft invalidates Brim and vice versa.
When Brim is selected it should be possible to select the Brim type in another drop down menu, for Raft it should remain faded.

[image]
[image]

DVSVIDEO, 2023-11-15:
[image]

Eldenroot, 2023-12-02:
@igiannakas maybe another pr

closed 2024-03-10 by github-actions[bot]

Fmstrat, 2025-07-08:
This should be reopened. Skirt and Brim are under `Others`, Raft should be there, too.