OrcaSlicer/OrcaSlicer #2745: "Bottom interface spacing" option has no effect/"Top interface spacing" changes bottom spacing

trankillity, 2023-11-15:
### OrcaSlicer Version

1.8.0-rc2

### OS version

Windows 11

### Printer

Generic Klipper (Ender 3 S1 Pro w/ Sonic Pad)

### How to reproduce

1. Load a [support interface test](https://www.printables.com/model/198420-support-test) model.
2. Enable supports/support interface
3. Set different values for `Top interface spacing` and `Bottom interface spacing`
4. Slice model

### Actual results

Both top and bottom interface spacing are the same (mirroring the `Top interface spacing` value). `Top interface spacing` affecting both top and bottom can be verified by changing only the `Top interface spacing` and re-slicing.

[image]
[image]

### Expected results

1. `Bottom interface spacing` should change the bottom interface spacing. It currently has no discernible effect on slicing.
2. `Top interface spacing` shouldn't change the bottom interface spacing.

### Project file & Debug log uploads

No project or log files.

### Checklist of files to include

- [ ] Log file
- [ ] Project file

Noisyfox, 2023-11-15:
Does this only happen to certain support types/styles?

trankillity, 2023-11-15:
I just checked, it happens in Normal Auto/Manual. Tried all different patterns and bottom layer counts too. Still the same across all of them.

Noisyfox, 2023-11-15:
Right so it doesn't affect tree supports?

trankillity, 2023-11-15:
I've never used tree supports. It looks like tree supports completely ignore any "Bottom" settings anyway because there's no interface, and no spacing on the bottom when using Tree supports where the support starts on the model.
[image]

SoftFever, 2023-11-16:
Hmm, the bottom interface spacing is not supported by normal/organic support currently.

trankillity, 2024-02-15:
It may be stale, but it's still a bug :(

AuT-PsycHo, 2024-05-20:
Hello I have the same problem (2.0.0) found this and now commenting so maybe it will come back

gaminglaboratory, 2024-05-29:
I have the same problem with Orcaslicer (2.0.0)

closed 2024-09-04 by github-actions[bot]

Stormwing01, 2024-09-22:
Tree supports completely ignore bottom interfaces and spacing. basically: Tree supports print directly on your model without any bottom spacing, fusing the supports to the model permanently.

crookz, 2024-10-14:
same situation here, i need the tree supports to have interfaces with the model, currently its impossible
[image]

sssoaring, 2024-10-23:
Recommenting - 2.2.0-beta2 still has this error. Tried with Normal Snug supports.