OrcaSlicer/OrcaSlicer #2747: Steam OS(Arch Linux) support

hogum, 2023-11-15:
Hopefully support for steam os will be added so we can do slicing on the steam deck :)

hogum, 2023-11-16:
> You can already install it from aur https://aur.archlinux.org/packages/orca-slicer

Thank you for your reply. However, Steam OS is not a complete Arch Linux, so its support for aur is not friendly. Hopefully there is a flatpak or AppImage that can support it.

Rucadi, 2023-11-24:
There is an AppImage, but at least I get shared library errors, seems like not everything is packaged.

m0d3rnX, 2023-12-30:
AppImage is a worthless format, please provide a Flatpak to make sure it runs on every distro.

watzon, 2024-01-16:
Bambu Studio finally has an official Flatpak available, might be worth seeing how easy it would be to port their Flatpak to Orca.

arpia49, 2024-04-01:
My understanding is flatpak could be available _soon_ as https://github.com/SoftFever/OrcaSlicer/issues/3949 In the meantime I use Bambu Studio.

closed 2024-07-09 by github-actions[bot]