OrcaSlicer/OrcaSlicer #2751: Please add http upload and thumbnail functionality for the Snapmaker J1 printer.

rodneyflowers, 2023-11-15:
When I use the Orca Slicer to generate gcode for the Snapmaker J1 I have to save the gcode locally and manually move the file to the J1.  The files also do not show a thumbnail on the J1 once they are copied to the printer.

Using CURA, I was able to upload to the Snapmaker J1 after slicing.  I was also able to add a post process to generate a thumbnail that shows on the J1 screen.  I like everything else about Orca Slicer better but this one issue is a bit tedious to work around.

I would appreciate the same thumbnail and upload functionality I have in Orca for my Klipper based printers.

Note:  As of installing 1.8 RC2, I am unable to use the Orca slicer with a Snapmaker J1 profile because it does not show up in the list.

closed 2024-02-22 by github-actions[bot]

amckimm, 2024-08-27:
I also seek a solution to thumbnail compatibility for the Snapmaker J1/J1S

sillyfrog, 2025-05-17:
Posting here in case others come across this like like did.

This page outlines a solution using Luban: https://wiki.snapmaker.com/en/third_party_software/orca_slicer_with_snapmaker_printers (I've not tried this)

But I'm using this: [sm2uploader](https://github.com/macdylan/sm2uploader ), with the following configuration:
- Start it on the command line with: `sm2uploader -host <real ip address> -octoprint :8844`
- Configure OrcaSlicer (Clicking on the WiFi symbol next to the selected printer at the top of the side bar) with:
  - Host Type: Octo/Klipper
  - Hostname, IP: `http://127.0.0.1:8844`
  - leave the rest of the fields blank

Then when the slicing is done, you can click "Print" in the top right on OrcaSlicer, and it'll upload (it will _not_ start printing), `sm2uploader` will also _automatically_ fix the gcode so it works with the Snapmaker (thumbnail image, meta-data etc)