OrcaSlicer/OrcaSlicer #2752: If a plate is named, use the plate name as the default filename for Export G-Code and Print

jwyse, 2023-11-15:
Current behavior:

`Export G-Code` and `Print` will use the filename of one of the objects (presumably the first object that was added to the plate?) as the basis for the filename to export or send to the printer.

Desired behavior:

If a plate has been named, use the plate name as the basis for the gcode filename to export or send to the printer.  If the user has not set the plate name, then either use the plate number (ex: `plate_28_ABS_2h51m.gcode`) or the first object name (current behavior) as the default gcode filename.

Example:

In the screenshot below, I have arranged several STL objects on a plate, and named the plate `Controller mounts`.  When I clicked Print, the default filename is based off one of the objects on the plate.  As this plate is part of a much bigger project (printing Voron Trident parts on a small Voron 0.2), it's valuable to be able to (1) combine multiple parts on a plate (2) keep track of what has been printed, by plate rather than by individual part.

[image]

Of course the default filename can be modified before exporting or sending to the printer, but it would be a smoother workflow if the generated name used the plate name instead of a single part on the plate.

niklasb22, 2023-11-16:
Nice idea. I'll have a look at it

jwyse, 2023-11-20:
This is perfect!  I was looking for a way to customize the filename format in Preferences, and somehow completely overlooked that setting in the "Others" area of print settings.  Thanks!

closed 2023-11-20 by jwyse

jwyse, 2023-11-20:
This formatting snippet will name the output with the plate name if it is specified, otherwise the model filename.  This is exactly what I was looking for, and I like this as a default setting so that I don't always have to name every plate.

`{if plate_name!=""}{plate_name}{else}{input_filename_base}{endif}`

for example:

```
{if plate_name!=""}{plate_name}{else}{input_filename_base}{endif}_{filament_type[initial_tool]}_{print_time}.gcode
```

dstulken, 2024-02-15:
The snippet above works great - is there any reason why something like this isn't the default behavior?

deimon777, 2024-04-16:
Hello, how are you?
Do you know where I can see the gcode placeholders to export?
I was able to set the nozzle temperature, and the fill. But not the bed temperature, and not the print speed.

DarkmageDesigns, 2026-03-27:
> This formatting snippet will name the output with the plate name if it is specified, otherwise the model filename. This is exactly what I was looking for, and I like this as a default setting so [...]

I don't know you but I love you.