OrcaSlicer/OrcaSlicer #2753: Extruder motor stops working in middle of print

3Dmapsantos, 2023-11-15:
### OrcaSlicer Version

1.8.0 rc2

### OS version

Win11

### Printer

Ender 3 with marlin 2.1.1

### How to reproduce

The extruder stop's working in the midle of the print.
I had this problem before in 1.7.0 because i didn't the "g-code flaver" to Marlin2; after correcting was working all the way for the same part. Cura also. Stepper motors are not too hot.
it might be software or the printer?

The Ender config that i'm using:
[image]

[image]

### Actual results

The extruder stop's but the printer continues working in the empty. It stops around the Z:11mm

### Expected results

the stepper motor shold work without problems

### Project file & Debug log uploads

[3112_Helical_eggs_dispenser_PLA_1d13h51m.zip](https://github.com/SoftFever/OrcaSlicer/files/13367952/3112_Helical_eggs_dispenser_PLA_1d13h51m.zip)
[debug_network_Mon_Nov_13_07_10_08.log](https://github.com/SoftFever/OrcaSlicer/files/13367964/debug_network_Mon_Nov_13_07_10_08.log)
[LEON3D BLTouch 0.4 nozzle.json](https://github.com/SoftFever/OrcaSlicer/files/13367972/LEON3D.BLTouch.0.4.nozzle.json)
[0.12mm @LEON3D.json](https://github.com/SoftFever/OrcaSlicer/files/13367973/0.12mm.%40LEON3D.json)

### Checklist of files to include

- [X] Log file
- [X] Project file

Ocraftyone, 2023-11-16:
Could you please provide the generated gcode file as well?

3Dmapsantos, 2023-11-16:
> Could you please provide the generated gcode file as well?

thank you for the quick response. it's the [3112_Helical_eggs_dispenser_PLA_1d13h51m.zip](https://github.com/SoftFever/OrcaSlicer/files/13367952/3112_Helical_eggs_dispenser_PLA_1d13h51m.zip)

Mr-bbp, 2023-11-16:
I have the same problem. The extruder stops working at a height of +-6mm (depends on gcode). I  attach  one of my gcodes with this problem.
[Cali-Dragon_v1_PLA_27m46s.gcode_3.59549 g weight.zip](https://github.com/SoftFever/OrcaSlicer/files/13381053/Cali-Dragon_v1_PLA_27m46s.gcode_3.59549.g.weight.zip)

3Dmapsantos, 2023-11-16:
[image]
[image]
all black parts, diferent gcodes generated in orcaslicer, it stops at the same level. the white one is on cura

Mr-bbp, 2023-11-17:
Still having the same issue with 1.80 final release. It seems as if the extruder stops extruding progressively. I attach another g.code and som photos. With Cura & PrusaSlicer no problems! My printer: Tronxy X5SA Pro, with Marlin.
[image]
[image]
[image]

3Dmapsantos, 2023-11-17:
> with Marlin.

Marlin legacy or Marlin 2?

Mr-bbp, 2023-11-17:
> > with Marlin.
> 
> Marlin legacy or Marlin 2?

Marlin 2.. I tried a temperature tower and retraction test and print well... but when i put my .stl... the issue returns... tried portable & installed version... tried new profiles... and nothing works..

ThrillerKnight, 2023-11-19:
I've been having the same issue in 1.8. The extruder stops at the same height each time but the rest of the printer continues. 
Creality CR-10v3 with CR-Touch and Marlin 2.1.2. I got one successful print out of 3 sliced files with 5 print attempts total. Not sure why the successful one worked.

kabodos, 2023-11-24:
I have a very rare error in version 1.8.0, when I select Marlin 2, when I level the bed before starting a print, since I have the leveling in the start gcode, the nozzle hits the bed, but if I select Marlin Legacy, this does not happen, in my case it is an Anet ET4 printer with a MKS Robin Nano V3 board.

3Dmapsantos, 2023-11-25:
I have been tweaking my marlin 2.1.2 firmware, one of the tweaks was activating the junction deviaction. now i have consecutive prints that surpasse the 11mm issue, but now the problem shited to miss aligned layers. and i found out this other report regarding the junction: https://github.com/SoftFever/OrcaSlicer/issues/2209

igiannakas, 2023-12-10:
Potential fix published above. Please test and report back

3Dmapsantos, 2023-12-14:
> I have been tweaking my marlin 2.1.2 firmware, one of the tweaks was desactivating the junction deviaction. now i have consecutive prints that surpasse the 11mm issue, but now the problem shited to miss aligned layers. and i found out this other report regarding the junction: #2209

Since i changed the firmware, this error stopped. So i will close this. But the junction option it still a good option to implement.
Best Regards,
Maps

closed 2023-12-14 by 3Dmapsantos