OrcaSlicer/OrcaSlicer #2765: Supports go through the object I am trying to print .

kilomanjaro4, 2023-11-16:
### OrcaSlicer Version

1.7.0 and 1.8.2rc2

### OS version

Windows 10

### Printer

Bambu X1C

### How to reproduce

Pull the file into the orca slicer then hit slice with tree supports slim enabled. This will cause some of the supports to go through the  print.

### Actual results

v

### Expected results

c

### Project file & Debug log uploads

[LP4_Fan_Adaptor.zip](https://github.com/SoftFever/OrcaSlicer/files/13384587/LP4_Fan_Adaptor.zip)

### Checklist of files to include

- [ ] Log file
- [X] Project file

Noisyfox, 2023-11-17:
Could you also upload the 3mf file?

kilomanjaro4, 2023-11-17:
[LP4_Fan_Adaptor3mf.zip](https://github.com/SoftFever/OrcaSlicer/files/13387482/LP4_Fan_Adaptor3mf.zip)

here it is

Noisyfox, 2023-11-17:
Turn on "build plate only" should solve the problem.

kilomanjaro4, 2023-11-17:
I'm so dumb. Thankyou. First time ever creating an issue and this happens. Sorry for wasting your time.

closed 2023-11-17 by kilomanjaro4

mnott, 2025-02-23:
I had it on "on build plate only", and for some reason, the support interface (which was PLA compared to the rest of the print in PETG) went through all objects. I'm trying again using no prime tower figuring maybe that's the problem. But he should not have done that.