OrcaSlicer/OrcaSlicer #2766: Crash to desktop when slicing certain models.

jcford73, 2023-11-16:
### OrcaSlicer Version

1.8.0-rc2

### OS version

Windows 11 Pro

### Additional system information

CPU: Intel Core i9 13900k
RAM: 64GB DDR5
GPU: GeForce RTX 4090

### Printer

Sovol SV06

### How to reproduce

Simply attempt to slice the model

### Actual results

Slicing certain models such as this grim reaper figure cause the program to crash to desktop with no error message. This happens in PrusaSlicer 2.6.1 as well. Likely affects the whole slic3r family.

I ran OrcaSlicer and PrusaSlicer in Ubuntu in a VirtualBox VM and that version was able to slice the models, if slowly.

Today, I tried turning on Windows 8 compatibility mode for OrcaSlicer in Windows 11. To my surprise, it didn't crash this time. It took a _very_ long time to slice the models, but it worked. So I have a workaround.

Feedback on Reddit indicated others have this problem with certain models in the Windows version of the software. Hopefully this Windows 8 compatibility clue will help diagnose the cause.

### Expected results

Successful slicing

### Project file & Debug log uploads

[crash_Thu_Nov_16_14_25_33_0.zip](https://github.com/SoftFever/OrcaSlicer/files/13384622/crash_Thu_Nov_16_14_25_33_0.zip)
[ReaperCloak_Main.zip](https://github.com/SoftFever/OrcaSlicer/files/13384625/ReaperCloak_Main.zip)

### Checklist of files to include

- [X] Log file
- [X] Project file

Noisyfox, 2023-11-17:
Could you try this build https://github.com/SoftFever/OrcaSlicer/actions/runs/6897958644
and upload the new crash log? We've improved the crash log a little bit and would like to see how effective it is. Thanks.

jcford73, 2023-11-17:
I tried that build. It does still crash, but it does not generate a crash log file at all. Here's the debug log file it generates, though.
[debug_Fri_Nov_17_13_51_24_18564.log.zip](https://github.com/SoftFever/OrcaSlicer/files/13397363/debug_Fri_Nov_17_13_51_24_18564.log.zip)

silence48, 2023-12-31:
Another c0000005 ACCESS_VIOLATION
Related issues include #2921 #3375 #3340 #3292 #3157 #2896

jcford73, 2024-02-22:
Any news on these crashes? 1.9.1 is still crashing on a lot of models. Sometimes regardless of settings, sometimes only when trying to generate tree supports. Always the same CTD behavior.

HellBz, 2024-03-12:
Same Problem, but my Slicer simply closing, without Crash-Log.

[debug_Tue_Mar_12_20_39_00_26352.log](https://github.com/SoftFever/OrcaSlicer/files/14577860/debug_Tue_Mar_12_20_39_00_26352.log)

jcford73, 2024-03-18:
In case it helps debug this issue, I've found if I set process priority to realtime and the processor affinity to only use the first half-dozen or so cores, it will slice. It will just take a really long time. This is probably similar to what is happening when I was turning on Windows 8 compatibility mode.

Noisyfox, 2024-03-19:
Is there any crash log available?

jcford73, 2024-03-21:
Here's the most recent using OrcaSlicer 2.0 beta. And here's the system specs again just for reference.

OS version
Windows 11 Pro

Additional system information
CPU: Intel Core i9 13900k
RAM: 64GB DDR5
GPU: GeForce RTX 4090

[debug_Tue_Mar_19_22_20_28_23524.log.zip](https://github.com/SoftFever/OrcaSlicer/files/14680367/debug_Tue_Mar_19_22_20_28_23524.log.zip)

jcford73, 2024-06-21:
This issue should not close. It's definitely still happening. I have to set processor affinity to limit the number of cores in use by the software or it will crash when slicing any but the simplest of models.

RoccoFL, 2024-07-15:
I also have an i9 13900K and am having issues, windows 8 compatibility seems to be helping. Note, I am having this issue on all slic3r-based slicers (slic3r, superslicer, and orcaslicer...I haven't gotten around to trying prusaslicer)

Kontra21, 2024-07-27:
I am also using an i9 13000KF and having issues slicing certain models with orca crashing. I tested orca after bambustudio.

closed 2024-11-03 by github-actions[bot]