OrcaSlicer/OrcaSlicer #2768: Can't export gcode file due to bad filename_format template using MK3S profile

Kytra, 2023-11-17:
### OrcaSlicer Version

1.8.0-RC2

### OS version

Windows 10

### Printer

Prusa MK3S

### How to reproduce

1. Choose "Prusa MK3S 0.4 nozzle" machine profile, or any profile that inherits it.
2. Add any model then click "Slice Plate".
3. Click "Export G-code file".

### Actual results

[image]

### Expected results

Show explorer to save the gcode file with proper file name.

### Project file & Debug log uploads

[Testing_Model.zip](https://github.com/SoftFever/OrcaSlicer/files/13388138/Testing_Model.zip)
[debug_Fri_Nov_17_14_14_46_10544.log.zip](https://github.com/SoftFever/OrcaSlicer/files/13388140/debug_Fri_Nov_17_14_14_46_10544.log.zip)

### Checklist of files to include

- [X] Log file
- [X] Project file

Kytra, 2023-11-17:
Already tested on 1.8.0-RC1, same problem. All the settings (Printer, Filament, Process, etc.) are default.

SoftFever, 2023-11-17:
Thanks for reporting
Fixed.

commit cb038d04 (2023-11-17): fix Prusa MK3S and Mini profile issues
Fixed #2768
closed 2023-11-17 by SoftFever