OrcaSlicer/OrcaSlicer #2770: X- Y- hole compensation

BahamutWolf, 2023-11-17:
I carried out the Orca tolerance test and set an X- Y_ hole compensation of 0.3. However, the problem with the hole compensation is that it does not fully compensate for holes. I have noticed an offset in 2 of my projects that I have designed with Fusion 360 that is caused by the compensation. You could solve the problem by using the X-Y contour compensation, but then the workpiece is no longer dimensionally accurate.

[image]
[image]
[image]
[image]

SoftFever, 2023-11-18:
It is because the hole compensation will stop at the non-hole area as the picture demonstrated.
We don't have a perfect solution for such cases, the only way I can think of is bringing the model back to CAD software and adjust the tolerance.

[image]

BahamutWolf, 2023-11-18:
I understand. I thought that was a bug. I measured the outer diameter of the sleeve and it is 0.3mm too wide, so the hole should fit due to the compensation of 0.3 and contour compensation at -0.3. This step is then gone in any case. Thank you very much.

closed 2023-11-18 by BahamutWolf