OrcaSlicer/OrcaSlicer #2773: "Arc fitting" making error artefacts in gcode

surikatec, 2023-11-17:
### OrcaSlicer Version

1.8.0

### OS version

Windows 11

### Additional system information

Ryzen 5 5700X
16GB RAM
1080Ti

### Printer

Custom / VzBot / Whatever

### How to reproduce

With allowed "Arc_fitting" makes randomly errors in G-code.

[image]
[image]
[image]
[image]

### Actual results

Turn off "Arc_fitting".

### Expected results

Repair it.

### Project file & Debug log uploads

...

### Checklist of files to include

- [ ] Log file
- [ ] Project file

surikatec, 2023-11-17:
Tried reinstall Orca, tried portable version. Still same error.

SoftFever, 2023-11-18:
can you attach the project?

surikatec, 2023-11-18:
> can you attach the project?

You mean .3mf file?

discip, 2023-11-21:
> You mean .3mf file?

Yes he does. 😊

closed 2024-02-28 by github-actions[bot]