OrcaSlicer/OrcaSlicer #2774: [Feature Request] Add option to remove very short perimeters

scottmudge, 2023-11-17:
I've already tried enabling and changing the `Filter Out Tiny Gaps` setting, but it has no effect on the annoyance of very short perimeters being put in places they don't really need to be.

For instance:

[image]

They're less than 1mm long, and the time it takes to retract and move between each of these tiny dots adds up over the course of a print.

It would be nice to remove/ignore very short perimeters which have no appreciable effect on overall strength, and which just waste print time.

killerline10168, 2023-11-18:
+1

They may also increase stringing, especially for bowden extruders. 
One might also argue that travel moves like these marginally increase printer wear, which adds up after many prints.

scottmudge, 2023-11-18:
I found that for at least in the Arachne engine, modifying line `WallToolPaths.cpp:674` from:

`            if (line.is_odd && !line.is_closed && shorterThan(line, min_width / 2)) { // remove line`

to something like

`            if (line.is_odd && !line.is_closed && shorterThan(line, min_width * 2)) { // remove line`

Removes the small segments. I'll try parameterizing it and adding a config field for it.

scottmudge, 2023-11-18:
Alright I added a new option to the Arachne settings. Looks like it works fine. Increasing the value removes longer and longer unclosed (non-loop) wall sections:

[image]

Adjusting from the default hard-coded value to 2 or 2.5 reduced a 12 minute print to just under 11 minutes, so saved over a minute (and countless retractions/moves).

scottmudge, 2023-11-18:
Alright made a final change to ensure the topmost layer uses the default value. Will create a PR soon.

scottmudge, 2023-11-18:
Final update, that test print in the screenshots printed far cleaner and with less stringing when those tiny walls/perimeters were removed. So I think the feature has some benefit for smaller features like in this test model.

scottmudge, 2023-11-18:
Alright PR created, #2790

closed 2023-11-18 by scottmudge

killerline10168, 2023-11-19:
> Alright PR created, #2790

Thank you for your service!

hugo-id, 2024-12-09:
Thanks for this feature!

I just switched from PS and this is an absolute game-changer!