OrcaSlicer/OrcaSlicer #2782: Black build plate

lin-ycv, 2023-11-18:
### OrcaSlicer Version

1.8.0

### OS version

Windows 11

### Printer

Ender 3 s1 pro

### How to reproduce

On a HP omen laptop, the build plate doesn't display properly for ender 3 s1 pro.
It shows up as a black sheet instead of the build plate graphics.
Other printer options display the build plate just fine. 

### Actual results

[image]

### Expected results

For the build plate to display properly. 

### Project file & Debug log uploads

Not a project or crash.
Tried both portable and installed version.
Only happens on this omen laptop. 

### Checklist of files to include

- [ ] Log file
- [ ] Project file

Noisyfox, 2023-11-18:
Is your laptop equipped with an AMD gpu?

lin-ycv, 2023-11-18:
It's a Ryzen CPU with Nvidia GPU.

Is there incompatibilities with AMD GPU? On my desktop it's AMD iGPU, it displays fine.

Noisyfox, 2023-11-18:
Do you mind giving this build a try and see if it helps? https://github.com/SoftFever/OrcaSlicer/actions/runs/6911265809

lin-ycv, 2023-11-18:
yes, this build has solved the issue.

closed 2023-11-18 by lin-ycv

Noisyfox, 2023-11-18:
Good to know that, thanks!

adolfotregosa, 2024-08-19:
__EGL_VENDOR_LIBRARY_FILENAMES=/usr/share/glvnd/egl_vendor.d/50_mesa.json ./OrcaSlicer_Linux_Ubuntu2404_V2.2.0-dev_c6d7155092af9b224253666b778b5e75.AppImage 

Does this help ???

PR #2780 (2023-11-19): Fix PNG build plate texture not rendering on AMD GPUs
This should fix #2713
Also fix #2782

cherry picked from commit prusa3d/PrusaSlicer@971f2a08e2758ea1d91e75af9b1443d14b41f895
cherry picked from commit prusa3d/PrusaSlicer@eee4453993ba38f3f11c8f89c56cefa50eaaf74b