OrcaSlicer/OrcaSlicer #2787: Flush volume and time incorrect without Prime Tower

zviratko, 2023-11-18:
### OrcaSlicer Version

1.8.0 (also on RC2)

### OS version

macOS Sonoma

### Printer

Bambu Lab X1C

### How to reproduce

1. Slice the included file
2. Look at filament consumption and time - it takes 9h and 80g of filament
3. Enable Prime Tower and reslice
4. Look at filament consumption and time - now it takes 13h30m and 270g of filament. When looking at Line Type chart, Prime tower supposedly takes 4h45m to print.

I opened the same project in Bambu Studio and there the difference is a few minutes and a few grams, just for the prime tower (and it's similiar to the higher number OrcaSlicer reports with Prime Tower enabled).

### Actual results

I believe OrcaSlicer ignores all flushing when Prime Tower is disabled (or color changes are broken, I have not tried printing it)

### Expected results

Difference between Prime Tower enabled and disabled should be much less, flushing should be calculated correctly and shouldn't differ from Bambu Studio that much.

### Project file & Debug log uploads

[cat it.zip](https://github.com/SoftFever/OrcaSlicer/files/13400341/cat.it.zip)

### Checklist of files to include

- [ ] Log file
- [X] Project file

foxmckraut, 2023-11-18:
[image]
cann confirm it.
[image]

AdrianGarside, 2023-11-19:
Came here to log this same issue. With the prime tower disabled, no purging/flushing occurs:

Orca Slicer 1.8.0:
[image]

Bambu Studio 1.7.6.92:
[image]

With prime tower enabled, Orca Slicer changes to:
[image]
Which now matches the results from Bambu Studio with/without prime tower enabled.

SillacSaurfang, 2023-11-19:
Can confirm, woke up to this this morning
[image]
Was not an issue in RC2.
Also, it seems if you're printing multiple different-colored objects (and by object) the purges between objects are fine.

tripp396, 2023-11-22:
Can confirm. Thought I was taking crazy pills until I watched a transition. It seems the first transition purged but not very much. Second transition where I setup a larger flushing volume didn't flush at all

dartrax, 2023-11-22:
This is a severe bug, making multi color printing basically impossible right now (without prime tower).
Obviously we have not tested the RC thoroughly enough ;-)
I have to downgrade until it's fixed.
If this is the case for every multicolor print without prime tower, I would even go so far as to recommend to withdraw the release, because it would generate many failed prints and therefore wasting filament and electricity.

ericmerrill, 2023-11-25:
I'm seeing the same problem with purges being completely wrong without a purge tower.
It's even a problem when printing objects one at a time in different colors - there is no purge between the objects.

yuyeshun, 2023-11-26:
I'm seeing the same problem too.
one object one at a time in different colors(without purge tower)

ericmerrill, 2023-11-26:
Pretty sure this was caused by [this commit](https://github.com/SoftFever/OrcaSlicer/commit/bbdd580bb146782e3a3bed4a2315fed962b9e5b9) added between 1.8.0-RC2 and 1.8.0 release by @SoftFever. I couldn't immediately tell if there is a Issue or PR to go with that commit though.

igiannakas, 2023-11-28:
Color change was broken. New release fixes that - can download it from the actions tab. Please retest and report back.

mmahon15906, 2023-11-28:
This looks to have resolved the issue for me. I did have my flush volume multiplier to higher than I normally would (didn't realize it was this way from my initial testing when there still was a problem), but even those high volumes did not work previously. I will be doing another print in about 2 hours back to my original settings, and assume my silence means its still working!

Noisyfox, 2023-11-29:
This should have been fixed now, we will release a hotfix version asap.

dartrax, 2023-12-03:
I confirm that this is fixed: https://github.com/SoftFever/OrcaSlicer/actions/runs/7018932845
Thanks for the fix!

closed 2024-03-12 by github-actions[bot]