OrcaSlicer/OrcaSlicer #2789: Login / Register button launches small empty window

karmakina, 2023-11-18:
### OrcaSlicer Version

1.8.0

### OS version

Windows 11 Home

### Additional system information

OS Microsoft Windows 11 Home
Version	10.0.22621 Build 22621	
System Type	x64-based PC
Processor	Intel(R) Core(TM) i7-8700K CPU @ 3.70GHz, 3696 Mhz, 6 Core(s), 12 Logical Processor(s)	
Graphics: NVIDIA GTX 1080 Ti
BIOS Version/Date	American Megatrends Inc. 1.A0, 6/8/2020	
Installed Physical Memory (RAM)	32.0 GB	
Total Physical Memory	31.9 GB	
Available Physical Memory	16.5 GB

### Printer

Bambu Labs X1C

### How to reproduce

Open Orca Slicer 1.8.0m, click Login/Register.

### Actual results

A small blank window appears and that's all. 

### Expected results

I should be able to log in, I presume. 

### Project file & Debug log uploads

There is nothing to upload, the software simply does not login. I only have one image as that's all there is. 

[image]

### Checklist of files to include

- [ ] Log file
- [ ] Project file

karmakina, 2023-11-18:
Unsure if it matters, but my default browser is Opera. However, Bambu slicer has no issues logging in via Opera, so I don't know why Orca would have trouble calling that browser.

MallocArray, 2023-11-22:
Duplicate of #2004 resolution is posted in there about how to get the network plugin files from Bambu Studio to copy into your Orca folder

Ocraftyone, 2023-11-22:
I am getting the same thing one the instance I have installed on my system, but the one in my dev environment seems to be functioning properly (Showing that the plugin is not installed and not allowing the button to be clicked).

closed 2024-02-29 by github-actions[bot]