OrcaSlicer/OrcaSlicer #2792: PA Pattern Test: Retracts in Solid Infill Area

CCS86, 2023-11-19:
### OrcaSlicer Version

1.8.0

### OS version

Win 10 x64

### Printer

X1C

### How to reproduce

Generate a PA pattern calibration.

### Actual results

Retract moves are generated for every tiny travel in the the solid infill area:

[image]

### Expected results

I expect unretracted travel here because they are shorter than the Travel Distance Threshold

### Project file & Debug log uploads

...

### Checklist of files to include

- [ ] Log file
- [ ] Project file

closed 2024-05-27 by github-actions[bot]

CCS86, 2024-05-29:
This seems like a really easy fix @SoftFever

SoftFever, 2024-05-29:
Have you tried disable this option?

[image]