OrcaSlicer/OrcaSlicer #2795: Settings seam placement to back creates unnecessarily long travel moves

killerline10168, 2023-11-19:
### OrcaSlicer Version

1.8.0

### OS version

Windows 11 23H2

### Printer

Ender 3 Max

### How to reproduce

1. Insert shelled out cube (see pics)
2. Set seam position to back
3. Watch where the seams are. 

### Actual results

The seams are on opposite sides at the back, allowing for lots of stringing, long travel moves and even ruining prints (just happened yesterday on a 230 gram print). This also creates more wear and a lot more noise on the printer compared to if the seams where closer together.

[image]

This is the ruined print because I forgot to check where the seams where created in Orca:

[image]

### Expected results

The seams should optimally end up as close to each other horizontally as possible, to reduce all the aforementioned effects, like the picture below:

[image]

### Project file & Debug log uploads

[Box.zip](https://github.com/SoftFever/OrcaSlicer/files/13402805/Box.zip)

Project file as well as settings used.

### Checklist of files to include

- [ ] Log file
- [X] Project file

killerline10168, 2023-11-19:
I should also include that setting the seam position to **Aligned** causes the same issue, but even worse:

[image]

The seams are now spread out across three corners and aren't even aligned (seam painting has been cleared ofc).

DVSVIDEO, 2023-11-19:
I don't think it's a mistake. It's a specific behaviour. The location of the seams is calculated differently in almost every model by the algorithm.
There are 4 kinds of automatism and if the result of these is not satisfactory, there is the possibility to place your own seam by painting.
It is not a mistake if you forget to check that you have placed it correctly by the algorithm.
Part of printing is adjusting these.
The Slicer cannot guess every user's idea.  The way it generates it is not always right for me either, so I paint when I need to.

I think delete this "error" there are enough actual errors.

killerline10168, 2023-11-19:
> I don't think it's a mistake. It's a specific behaviour. The location of the seams is calculated differently in almost every model by the algorithm. There are 4 kinds of automatism and if the result [...]

First of all, can you state the benefits of the way the seams are placed in my screenshots? If you even bothered to read my ticket, I gave great reasons as to why they **shouldn't** be calculated and placed this way. You can also see in my print that the algorithm is flawed, as it ruined my print. Sure, I could've saved it by manually painting the seams, but it should not ruin the print by default. I would argue that making the software more automatic is a great way to reduce spent time slicing files and attract more users. 
  
The TL:DR of my ticket is that the algorithm should **prefer and favor** short travel moves, which otherwise (once again), create potential for stringing, lots of noise and unnecessary retractions, wear on the printer, increased print times, less chance of print failure, etc. 

Also, how is **this** not an error:

[image]

They're clearly not aligned correctly, but that was the least of the problems reported here. 

I feel like you should dial back the passive aggressiveness, as we're both here to support and improve this software the best way we know how.

MrDixioner, 2023-11-20:
I fully support it! There should be a separate setting responsible for matching the seams so that they are located as close to each other as possible, if this cannot be done by the program algorithm, which at the moment works simply badly. Yes, you can place the seams "Back" or draw them manually, but there should always be fallback options so that the program's algorithm places the seams in one place, either automatically or manually using the settings.

killerline10168, 2023-11-20:
> I fully support it! There should be a separate setting responsible for matching the seams so that they are located as close to each other as possible, if this cannot be done by the program [...]

Thank you! However, wouldn't it be always better for the seams to be as close as possible, or are there scenarios where we wouldn't want that? I am all for more settings, but SuperSlicer is really overwhelming because of all its settings. 
If there are benefits to having it choosable, then great! Otherwise, I would say bake it in the algorithm. 

What are your, and everyone else's thoughts?

DVSVIDEO, 2023-11-20:
VI took the time to download your project file. That's why I wrote that I don't understand your problem:

I didn't touch anything, I just sliced and diced the original project.
The seam is the same inside and outside. it's perfectly straight, not like yours:
[image]
[image]

And travel is minimal:
[image]

I still maintain that developers have to deal with problems that the user cannot solve in any way. There is enough of that... Let's not make everything a problem.
If that is passive aggressiveness then I'll take it.

DVSVIDEO, 2023-11-20:
I agree that we disagree.
I started 3D printing 7 years ago, when there was only Cura and Simplify3D. There was no Creality company either.
My printer is a Micromake C1 that I completely rebuilt. Linar sinek, Voron Stealthburner, Core XZ kinematics, 300x300 bed, clipper firmware.
I've solved many, many problems over the years and watched the slicer programs evolve. Today they solve almost everything and still not enough...
They cannot guess our thoughts. As many people, as many prints, as many needs. It can't all be solved by one slicer.
If you have to create the seams by hand, create them by hand. You're doing yourself a favour if you look carefully at the model you're printing and set it up before you print.
As much as I'd like to achieve this, the complexity of 3D printing means it will never be a fully automated, grab and drop, and done process.
Not everyone needs to 3D print at any cost. It will always be a hobby for problem solvers and tinkerers.

closed 2024-03-14 by github-actions[bot]