OrcaSlicer/OrcaSlicer #2796: Seam painted line is skew on certain flat surfaces

killerline10168, 2023-11-19:
### OrcaSlicer Version

1.8.0

### OS version

Windows 11 23H2

### Printer

Ender 3 Max

### How to reproduce

1. Insert model.
2. Click seam painting, enable vertical.
3. Draw on a flat surface
4. A skewed line appears

### Actual results

[image]

This should be a bug, as it definitely isn't vertical to the bed, and for some reason, it only happens on the inner surface.

### Expected results

It should look like this (this is an outer surface):

[image]

### Project file & Debug log uploads

[Box.zip](https://github.com/SoftFever/OrcaSlicer/files/13402836/Box.zip)

### Checklist of files to include

- [ ] Log file
- [X] Project file

DVSVIDEO, 2023-11-19:
I copied your photo and drew a line from the edge of the picture to the painted seam.
I copied this line and pasted it to the bottom half of the seam.
The seam painting is perfectly vertical. The mistake is yours. The model is not vertical.

[image]

This "error" is not an error...

killerline10168, 2023-11-19:
I see you're here as well... Let me make it even clearer for you:

[image]

[image]

From both these angles, you can clearly see that the line is skewed. How about you contribute by either opening the attached project file or a file of your own, to confirm if this is exclusive to me or others as well?

DVSVIDEO, 2023-11-20:
1. In the slicer, you can rotate the slide in any direction, but always exactly horizontal
2. The photo you take with screen capture is always a perfect right-angled square
3. The "vertical" stitching option will only produce a vertical stitch line

It follows: if the seam is perfectly vertical and the subject plate (bed) is always vertical and the screen shot is always vertical and right-angled, then all points of the seam are perfectly equidistant from the edge of the image (regardless of the shape of the model)
I have already given proof of this, but here it is:
[image]

DVSVIDEO, 2023-11-20:
By the way, have you tried turning on the orthogonal view before painting?
File menu right from the drop down menu view----> use orthogonal view.
And then you'll realize it's not a bug...

killerline10168, 2023-11-20:
> By the way, have you tried turning on the orthogonal view before painting? File menu right from the drop down menu view----> use orthogonal view. And then you'll realize it's not a bug...

Damn, this was it. Great job figuring it out, thank you for taking your time and finding the issue. I'll close this as solved!

closed 2023-11-20 by killerline10168