OrcaSlicer/OrcaSlicer #2797: First layer travel acceleration

shyblower, 2023-11-19:
**Is your feature request related to a problem? Please describe.**
To print very soft TPU (60A, 70A) successfully (even on a Bambu Lab printer) you usually need very low acceleration values (around 150mm/s²) to keep the extruder from clogging but still be able maintain a decent maximum volumetric speed (around 3-4mm³/s). These low acceleration values should only apply to print moves and not for travel as it would lead to excessive stringing. This works on all layers but the bottom one where there is only one first layer acceleration setting which overrules travel acceleration.

**Which printers will be beneficial to this feature**
I suppose every printer

**Describe the solution you'd like**
I'd like to have a separate first layer travel acceleration setting to keep print acceleration for the first layer low but be able to configure first layer travel acceleration independently

**Describe alternatives you've considered**
None that could achieve the intended goal.

DVSVIDEO, 2023-11-19:
Take a closer look:
[image]

Then delete this error mark...

shyblower, 2023-11-19:
I was not talking about travel speed but about travel acceleration!

DVSVIDEO, 2023-11-19:
Yes! I am wrong, sorry...

closed 2024-02-26 by github-actions[bot]

Z18Aplha, 2024-03-05:
I would be interested in this as well!

vgdh, 2024-03-31:
Absolutely necessary!