OrcaSlicer/OrcaSlicer #2801: Calibration: PA Lines Forced Low Accel

CCS86, 2023-11-19:
### OrcaSlicer Version

1.8.0

### OS version

Win 10 x64

### Printer

X1C

### How to reproduce

When using the "lines" PA calibration, there is a forced acceleration value of 500 mm/s2 in the gcode. This makes the test pretty useless.

### Actual results

[image]

### Expected results

The last given M204 should match the "first layer" acceleration value defined in the slicer.

### Project file & Debug log uploads

...

### Checklist of files to include

- [ ] Log file
- [ ] Project file

capsel22, 2024-02-22:
I too would like to bump

closed 2024-05-30 by github-actions[bot]

CCS86, 2024-05-31:
@SoftFever

SoftFever, 2024-05-31:
It just respects your print setting.
500 is for first layer, from second layer it uses 5000 accel
[image]

CCS86, 2024-06-01:
> It just respects your print setting. 500 is for first layer, from second layer it uses 5000 accel 

The "lines" PA test is only one layer high. 

Previously, with first layer and wall accels set to 5000, it was still asserting an M204 S500

I just tested in the new 2.1.0 beta and it appears to have been resolved at some point.