OrcaSlicer/OrcaSlicer #2803: Crash to desktop when trying to open an incompatible 3mf project file

mmoraes80, 2023-11-19:
### OrcaSlicer Version

1.8.0

### OS version

Windows 10 64-bit

### Additional system information

CPU: Intel(R) Core(TM) i7-7700HQ CPU @ 2.80GHz   2.80 GHz
Memory: 16.0 GB (15.9 GB usable)
GPU: GTX 1060 Q-Max Design / Intel HD Graphics 630

### Printer

Original Prusa i3 MK3S (0.4 Nozzle)

### How to reproduce

1. Open "MK4 Rocket_Engine.3mf" from Printables
2. System shows a warning message "The 3mf is not supported by OrcaSlicer, loading geometry data only"
3. Click OK
4. Model is briefly loaded into plate
5. System crashes to desktop

### Actual results

Crash to desktop without any warning message.

### Expected results

Complete loading of geometry data into plate

### Project file & Debug log uploads

[Rocket_Engine.zip](https://github.com/SoftFever/OrcaSlicer/files/13404337/Rocket_Engine.zip)

### Checklist of files to include

- [ ] Log file
- [ ] Project file

fuzelet, 2023-11-21:
I am experiencing this exact same issue however I am using another model.  Adding my feedback in hopes it helps or if anything else I can do to help.  I am on Windows 11 23H2.

OrcaSlicer v1.8.0

model:
https://www.printables.com/model/574747-strong-flying-propeller-remix-v2

logs:
[log.zip](https://github.com/SoftFever/OrcaSlicer/files/13431943/log.zip)

reynol3232, 2023-12-02:
Same issue

silence48, 2023-12-31:
> I am experiencing this exact same issue however I am using another model. Adding my feedback in hopes it helps or if anything else I can do to help. I am on Windows 11 23H2...
> logs: [log.zip](https://github.com/SoftFever/OrcaSlicer/files/13431943/log.zip)

Another Exception Code :c0000005 ACCESS_VIOLATION
Likely related to issues the following issues:
[2921](https://github.com/SoftFever/OrcaSlicer/issues/2921) 
[3375](https://github.com/SoftFever/OrcaSlicer/issues/3375) 
[3157](https://github.com/SoftFever/OrcaSlicer/issues/3157)
[3340](https://github.com/SoftFever/OrcaSlicer/issues/3340)
[3292](https://github.com/SoftFever/OrcaSlicer/issues/3292)
[2896](https://github.com/SoftFever/OrcaSlicer/issues/2896)
[2766](https://github.com/SoftFever/OrcaSlicer/issues/2766)
[2888](https://github.com/SoftFever/OrcaSlicer/issues/2888)
[3174](https://github.com/SoftFever/OrcaSlicer/issues/3174)
[2803](https://github.com/SoftFever/OrcaSlicer/issues/2803)

closed 2024-04-07 by github-actions[bot]