OrcaSlicer/OrcaSlicer #2805: Retract not shown in Preview, possibly no Retract happening at all

Kraligor, 2023-11-19:
### OrcaSlicer Version

1.8.0-beta2

### OS version

Windows 10

### Printer

Anycubic Vyper

### How to reproduce

Enable retraction
Slice object
In Preview, select 'Line Type' and hide all line types except 'Retract' and 'Unretract'

### Actual results

Only Unretractions are shown, there is only one single Retraction visible, and it is at the purge line (see screenshot).
The printed model has lots of stringing, making me think there might not be any retraction happening at all.

[image]

### Expected results

For each Unretraction there should be a corresponding Retraction visible.

### Project file & Debug log uploads

[PowerMod_v3_testpiece.zip](https://github.com/SoftFever/OrcaSlicer/files/13404804/PowerMod_v3_testpiece.zip)
[gcode.zip](https://github.com/SoftFever/OrcaSlicer/files/13404810/gcode.zip)

### Checklist of files to include

- [ ] Log file
- [X] Project file

SoftFever, 2023-11-20:
Hi,
Thanks for report the issue.

There is a bug when `Retract amount before wipe` is set to 0.
This will be fixed soon. 
As a workaround for now, we can avoid setting it to 0.
[image]

killerline10168, 2023-11-20:
I reported the same issue a while back but with no response. I will refer to this ticket there.

Kraligor, 2023-11-20:
Workaround: Set `Retract amount before wipe` to > 0
Will be fixed soon per @SoftFever

closed 2023-11-20 by Kraligor