OrcaSlicer/OrcaSlicer #2806: Support: Independent support layer height option missing

Billiam, 2023-11-19:
### OrcaSlicer Version

1.8.0

### OS version

Windows 10

### Printer

Bambu Lab P1P

### How to reproduce

1. Open support tab
2. Observe `Independent support layer height` is not visible
3. Search for `Independent` in options search, and select the `Advanced: Independent support layer height`
4. The tab changes to `Support` (if it wasn't visible).
5. Support layer height option still not visible

### Actual results

Independent support layer height option is not visible

### Expected results

Independent support layer height option is visible

### Project file & Debug log uploads

[Untitled.zip](https://github.com/SoftFever/OrcaSlicer/files/13404949/Untitled.zip)

### Checklist of files to include

- [ ] Log file
- [X] Project file

Noisyfox, 2023-11-20:
Isn't it at the last of the Advanced section?
[image]

Noisyfox, 2023-11-20:
You have to disable prime tower first.

Billiam, 2023-11-20:
> You have to disable prime tower first.

Thank you, that was it. Perhaps the option could be disabled rather than hidden when necessary.

The interaction of searching for an option, finding it, and then not being able to see it is unpleasant.

Noisyfox, 2023-11-20:
Yeah, I've raised #2170 for improving these but haven't had much progress yet.

closed 2023-11-20 by Billiam