OrcaSlicer/OrcaSlicer #2807: Mouse-ear brim: Allow block modifier or port Cura logic

R4toLoko, 2023-11-20:
The last time I used mouse-ear brim was ages ago with Cura.

You could either set it for the whole model or by selecting the areas, by clicking on a area/corner a mouse-ear brim was automatically added to that spot only.
I am trying that with Orca and mouse-ear brim is added to the whole model, that wouldn't be a problem if you could use block modifier to disabled the brim but that is not possible either.

It would be awesome if Cura way of doing this is ported, you can add it only on the spots that might give you headache.

[image]

No need to add brim there

[image]

Eldenroot, 2023-12-02:
@Noisyfox mouse ears were implented by you?

hermitdafunk, 2024-07-11:
+1 for enabling feature to be able to paint-on /block brim-mouse-ears. Certain models, such as hinges are filled with brim when you don't want them to, and only need a few attachment brims to keep the model on the build plate.

billbonney, 2024-09-03:
+1 👍

I have a similar issue where I want to use a brim to help with layer adhesion and stop warping. But the brim is being added where I don't want it. If you could add a modifier to suppress areas (like tree support suppression) that would be awesome.

You can see in the screen grab, outer brim (dark blue) is being added underneath where I don't want it (or need it) for my print.

[image]

TrentBartlem, 2024-09-03:
> +1 👍
>
> I have a similar issue where I want to use a brim to help with layer adhesion and stop warping. But the brim is being added where I don't want it. If you could add a modifier to suppress areas [...]

I believe that was raised previously, as #5669

billbonney, 2024-09-06:
@TrentBartlem #5669 is related to "Mouse Ear Outer Only" which is similar request to control brim, but it's automatic. Using modifier gives ultimate control to exclude/include problem areas

hermitdafunk, 2024-12-09:
How is this not a feature yet?

bump

MJ3Ds, 2025-01-04:
6

Hecatron, 2025-06-21:
Run into this problem myself recently since mouse ears are a lot better than brims especially for ASA
but I had an inside area that was getting mouse eared up that I wanted to avoid.
One way is if under Add Modifier, the modifier cube or shape you could have options for the brim type there

miketanct, 2025-12-06:
Please implement modifier that allows blocking brims!  How else can I remove the unwanted brims for the letters A & D? 

[image]