OrcaSlicer/OrcaSlicer #2809: Infill Path Planning Very Erratic

CCS86, 2023-11-20:
### OrcaSlicer Version

1.8.0

### OS version

Win 10 x64

### Printer

X1C

### How to reproduce

Observe project file's infill ordering.

### Actual results

Very fragmented ordering of infill lines

https://github.com/SoftFever/OrcaSlicer/assets/12439543/23603984-0873-43d3-8d7a-2a9ed52fc982

### Expected results

Infill would ideally start at once side and work continuously across the part.

### Project file & Debug log uploads

[Simple Indoor Glider V2.zip](https://github.com/SoftFever/OrcaSlicer/files/13405617/Simple.Indoor.Glider.V2.zip)

### Checklist of files to include

- [ ] Log file
- [X] Project file

DVSVIDEO, 2023-11-20:
The project in the video is not the one you linked.
However, you can also see that it continues to fill solid quite randomly.

igiannakas, 2024-01-11:
Issue doesnt seem to occur in the latest release. Can you check please?

https://github.com/SoftFever/OrcaSlicer/assets/59056762/315f1986-2d6b-4e97-90ca-d1c6cf1ff825

closed 2024-04-19 by github-actions[bot]