OrcaSlicer/OrcaSlicer #2810: First Layer Infill Overlap Wrong

CCS86, 2023-11-20:
### OrcaSlicer Version

1.8.0

### OS version

Win 10 x64

### Printer

X1C

### How to reproduce

Open attached project and look at the first layer. 

### Actual results

You will see that despite having a uniform line width between perimeters and infill, the stepover for each path type is different. 

The perimeter stepover is correct, but the infill stepover is too small, causing over-extrusion.

These two green lines are identical length. One of them covers 12 perimeters, the other covers 14 infill lines. This results in about 16% over-extrusion:

[image]

### Expected results

I expect the stepovers to be equal for equal line widths.

### Project file & Debug log uploads

[Infill overlap.zip](https://github.com/SoftFever/OrcaSlicer/files/13406175/Infill.overlap.zip)

### Checklist of files to include

- [ ] Log file
- [X] Project file

killerline10168, 2023-11-20:
Interesting. The file you are showing in your pictures looks like a great example. Could you perhaps print it out, to see if there is a noticeable difference?

CCS86, 2023-11-20:
I noticed the issue through printing, where the first layer perimeters were
perfect and the the infill was badly over extruded. Subsequent layers
improved.

Then I swapped out my print model for a primitive cube to see if the
behavior stayed the same, and uploaded it as the example.

CCS86, 2023-11-20:
More info: it only seems to be the case when "bottom shell layers" is set to zero.

igiannakas, 2024-01-11:
This appears as resolved in the latest release:

[image]

[image]

[image]

[image]

closed 2024-04-19 by github-actions[bot]