OrcaSlicer/OrcaSlicer #2811: Adjust the position of CONFIG_BLOCK in gcode file for FlashForge Printers

FlashforgeOfficial, 2023-11-20:
****Is your feature request related to a problem? Please describe.****
Re: Lately I noticed when generating gcode files for Flashforge printers in Orca,  the default CONFIG_BLOCK lines are at the end of gcode files, while for other printers the BLOCK is at the beginning.                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                            
****Which printers will be beneficial to this feature****
Re: Flashforge printers.

****Describe the solution you'd like****
Re: It would be convenient if the Orca slicer could help solve this problem with this feature request,  generating gcode files for Flashforge printers with the CONFIG_BLOCK lines at the first few lines of gcode like other printer. 

****Describe alternatives you've considered****
Maybe there is other solution that hasn't come to my mind.  Hi guys, @SoftFever @mariuske maybe any other suggestions?

****Additional context****
Re: I put some screenshots here for comparison CONFIG_BLOCK position in gcode for different printers. 
Figure 1. Bambu gcode for comparison
[image]
Figure 2. Flashforge gcode for comparison
[image]

Thanks!

SoftFever, 2023-11-20:
Typically, it is conventional and more readable to place the config block at the end of the G-code. 
This approach is also a standard practice, as adopted by PrusaSlicer.
Therefore, most printers' G-code files have the config block at the end.
To the best of my knowledge, firmware like Marlin, RRF, and Klipper all handle this well.

If you're encountering any issues with the config block's placement at the end, please describe the specific problems instead?

FlashforgeOfficial, 2023-11-20:
Hi @SoftFever! The preview of gcode files from Orca won't be displayed normally, and looks like pixel contour. Besides the estimated printing time is not available for Flashforge printers.
These bugs might be related with the order of the config block in case where the gcode file is large, and the printer is taking some effort in loading the config block in the end of file, and the printing preview is not loaded correctly.  FYI @SoftFever @mariuske, this issue #2677 for adding remaining time might be somehow related to order of the config block.

FlashforgeOfficial, 2024-02-19:
> Hi @SoftFever! The preview of gcode files from Orca won't be displayed normally, and looks like pixel contour. Besides the estimated printing time is not available for Flashforge printers. These [...]

@SoftFever @mariuske Hi all.  As the preview of slicer file and estimated remaining printing time are still not supported for Flashforge ad5m series. Still, I wonder if we could move the CONFIG_BLOCK to the start of gcode like Bambulab for Flashforge Printers experimentally, for a quick check.  If nothing helps or becomes worse we could roll back to end of gcode. Let me know your thoughts, thanks!

closed 2024-05-27 by github-actions[bot]