OrcaSlicer/OrcaSlicer #2818: Qidi xmax3 wifi problems with Orcaslicer

HTraff, 2023-11-21:
A similar problem was covered in case **#1005** for the qidi x smart 3, and I have followed the advice in that topic but no success. 
I don't know if these two printers are identical in wifi connectivity.

I experience problems to connect to my xmax3 and I haven't been able to fix it. I get the error message that octoprint isn't installed when trying to connect with the octo/clipper option in orca. Tried all the suggested port options available in the dropdown list in orcaslicer, but can't find the Fluidd option that seem to be a thing and mentioned in some places. 

The very annoying part is that when I unpacked the printer I was able to connect to it after some severe headaches (this was some week ago) and then when I wanted to pick this up again I couldn't connect. I dont remeber how it happened but I think it got into to the klipper environment to control the machine through the device tab. (not by trying to connect directly but had naturally entered the ip etc). I have tested to reach the printer from two different computers on the network, one wifi connected and one wired. Pinging the printer from either does not give a reply.

I have tried updating firmware, restarted the printer. Also tried to connect from simplify 3d v5 without success.

sondbite64, 2023-11-24:
I am having the same issue and have been struggling. I am hopeful to have access to the calibration tools in Orca Slicer to dial in filaments.

TheHookworks, 2023-12-04:
Sorry to get Htraff and Sondbite64's hopes up with a notification, but same issue here. Attempting to connect the Qidi Xmax3 to Orcaslicer via the Octo/Klipper host option only results in a rejection error. No other option mentions Klipper, but as Htraff said the Qidi slicer host title is Klipper (Via Moonraker). Is some other support needed or is there just an additional step that we're missing?

sondbite64, 2023-12-12:
I have been able to connect by adding *:10088* to the end of my IP address.

HTraff, 2023-12-12:
I have also been able to occasionally connect via a browser to the right IP with that port extension, but its extremely unreliable. Loses connection almost instantly and sometimes will not connect at all. The wifi is very strong where the printer is located so it's not that. Other printers like bambu p1s have 0 issues. For now I have been using usb-sticks on the qidi printers.

HTraff, 2023-12-15:
I tried some more things, and discovered the connection works WHILE I am starting up ubiquiti unifi network server. Then when its started and I can browse the network in a browser the connection is dead again. Seems to be the same for connecting with the browser to the printer. I know too little about networks obviously. :( Could ports be blocked on the internal network?

sondbite64, 2023-12-18:
Sorry I am not a network expert. I can say that the addition of *:10088* to
the end of my IP address has been working well for me.

closed 2024-03-25 by github-actions[bot]

GrimOutlook, 2024-09-11:
I came upon this thread trying to set up my Qidi X-Max 3D and the solutions in here worked for me. Simply add `:10088` to the end of the IP address of your printer and mine successfully connected using Orca with the Klipper printer dropdown item selected and `Ignore Untrusted Certificates` checked. [This article](https://wiki.qidi3d.com/en/software/ConnectToPrinterViaWIFI#remote-printing) has more information if needed.