OrcaSlicer/OrcaSlicer #2822: Slicing only produces a single bottom and top layer

e-dreyer, 2023-11-21:
### OrcaSlicer Version

1.8.0

### OS version

Pop!_OS 22.04 LTS

### Additional system information

CPU: Intel Core it-7300HQ 2.50GHz
Graphics: Nvidia MX150
Memory: 16GB DDR4
FreeSpace: 500GB+

### Printer

Creality K1

### How to reproduce

Slice the provided file

### Actual results

Slicing the provided file only produces a single top and bottom layer, although 5 top and bottom layers are specified as well as a minimum shell thickness

### Expected results

5 Top and bottom layers or at least a thickness of 0.8mm

### Project file & Debug log uploads

[construction.zip](https://github.com/SoftFever/OrcaSlicer/files/13425869/construction.zip)

### Checklist of files to include

- [ ] Log file
- [X] Project file

e-dreyer, 2023-11-21:
Slicing with 1.8.0-rc yields the correct results:
[construction_1_8_0_rc.zip](https://github.com/SoftFever/OrcaSlicer/files/13425888/construction_1_8_0_rc.zip)

pacmac, 2023-11-25:
I think I have the same issue, I can see the infill showing through on the top layer, it looks like there's only one layer on the top surface despite the fact that it's set to the default 3 layers.

[image]
[ReolinkCompoundAngles v1.stl.zip](https://github.com/SoftFever/OrcaSlicer/files/13463372/ReolinkCompoundAngles.v1.stl.zip)

e-dreyer, 2023-11-25:
> I think I have the same issue, but only the top layer, I can see the infill showing through on the top layer, it looks like there's only one layer on the top surface despite the fact that it's set to the default 3 layers.
> 
> [image]
> [ReolinkCompoundAngles v1.stl.zip](https://github.com/SoftFever/OrcaSlicer/files/13463372/ReolinkCompoundAngles.v1.stl.zip)
> 

You should only be able to see a single layer in the "Line type" view. Have you tried with 1.8.0-rc to ensure it isn't the model? Also try to set a minimum shell thickness. Otherwise, we might have found a bug.

pacmac, 2023-11-25:
Thanks, strange thing is the line type shows no top layer (red).
[image]

pacmac, 2023-11-25:
I just tried on V1.7, same problem, it does not appear to recognise the compound angled upper surface as the top surface.

e-dreyer, 2023-11-25:
Yeah, it is showing a print time as less than a second. I had similar results, but earlier versions worked for me. So your situation might be the model. Will open it on my side so see what I get.

pacmac, 2023-11-25:
Also tried PruSlicer and same issue, then tried Cura and Cura does recognise the top layer correctly, so I guess this is inherited from the original Slic3R.

closed 2024-03-02 by github-actions[bot]

pacmac, 2024-03-02:
It not inactive, it is just that nobody has responded to it :-)