OrcaSlicer/OrcaSlicer #2830: Filament library - not working

Eldenroot, 2023-11-21:
### OrcaSlicer Version

1.8.0final

### OS version

Win11 64bit

### Additional system information

---

### Printer

Bambu Lab A1 mini

### How to reproduce

Go to filament settings (presents), uncheck (deselect) the filaments brands which you do not wanna.  

### Actual results

[image]

Polylite, eSun, Polyterra are still displayed, even when you dint want to...

### Expected results

Unchecked filaments should not be displayed.

### Project file & Debug log uploads

---

### Checklist of files to include

- [ ] Log file
- [ ] Project file

Eldenroot, 2023-12-20:
@SoftFever - this one still happens

Even when you uncheck "Polyterra" and "eSun" - they are still displayed in the menu for filament selection:

[image]

[image]

SoftFever, 2023-12-20:
> @SoftFever - this one still happens
>
> Even when you uncheck "Polyterra" and "eSun" - they are still displayed in the menu for filament selection: [...]
Can you try the latest dev build?
I can't repro it with 1.9-dev
[image]

kylek29, 2023-12-20:
@Eldenroot Out of curiosity (for testing), do you have BambuLab cloud sync turned on or off?

Eldenroot, 2023-12-21:
@kylek29 @SoftFever 

I can still reproduce on latest Github nightly build (21122023). 

These filaments are unchecked, but they are still in the list. It does NOT happens with Bambu Studio slicer. 

[image]

I have disabled synchronize user settings in advance settings.

How to fix this, should I delete some folder with settings in my PC?

closed 2024-01-20 by Eldenroot