OrcaSlicer/OrcaSlicer #2834: Add Internal edge to internal surface areas

TheAtomicSoul, 2023-11-22:
**Is your feature request related to a problem? Please describe.**
I've noticed a lot of noise while printing and realized it's because the internal edge of surfaces is "floating" and has nothing to attach to. 
In comparison, there are edges for this in Cura. 

**Which printers will be beneficial to this feature**
all

**Describe the solution you'd like**
Add at least one internal edge for flat areas inside the model. 

[image]

[image]

What I get in Cura: 
[image]

TheAtomicSoul, 2023-11-22:
As visible in Cura, the bottom right area where the horizontal infill buts against an edge inside the model. 
Orca, meanwhile, leaves these edges floating in the air, with nothing to help them hold in place.

SoftFever, 2023-11-23:
What version of Orca are you using?
This has been greatly improved in Orca 1.8

discip, 2023-11-23:
@TheAtomicSoul
Have you already tested what was proposed to you?
Have you tried the latest version of OrcaSlicer?

Please remember closing your issue if it is solved. 😊

thanks

TheAtomicSoul, 2023-11-25:
> What version of Orca are you using? This has been greatly improved in Orca 1.8
Unless there's an option I don't know of to add it, It still lacks any internal edge to butt against, which can increase edges curling.

1.8.0 also makes random sections into concentric surface patterns. 

[image]

[image]

TheAtomicSoul, 2023-11-25:
> @TheAtomicSoul Have you already tested what was proposed to you? Have you tried the latest version of OrcaSlicer?

Nothing has been proposed though? And this is in 1.8.0 RC.

closed 2024-03-12 by github-actions[bot]

henrivdr, 2024-07-18:
@TheAtomicSoul was this fixed?

TheAtomicSoul, 2024-07-18:
> @TheAtomicSoul was this fixed?

Nope. 
[image]

TheAtomicSoul, 2024-07-18:
To clarify: the weird loops that I was getting was fixed, but the original request for an internal edge loop to support the internal infill wasn't.