OrcaSlicer/OrcaSlicer #2840: Thin walls with arachne have tiny moves causing blemishes

sethml, 2023-11-22:
### OrcaSlicer Version

1.8.0

### OS version

macOS 14.1.1 (23B81) arm64

### Printer

Elegoo Neptune 3 0.4mm nozzle

### How to reproduce

1. Load or create a model with curved thin walls about the nozzle thickness. Here's a half-sphere with 0.4mm wall thickness created with a sphere primitive and a concentric sphere primitive of 0.8mm smaller diameter subtracted:
[Sphere-Shell.3mf.zip](https://github.com/SoftFever/OrcaSlicer/files/13432610/Sphere-Shell.3mf.zip)
2. Slice and print the model.

### Actual results

The G-code for the curved thin walls has G1 segments with almost zero movement. On my Klipper v0.11.0-87 printer, this results in decelerations at each nearly-zero-movement move, which in turn result in small overextrusion blobs. From the comments embedded in the g-code, it appears that Arachne thinks the extrusion width should change a tiny amount for a tiny distance, for example from layer 30 of the attached Sphere-shell example, the exported g-code lines 3106-3113 (layer 30 operations 5-10) are (with comments added for move distance):

```
G1 X61.469 Y118.573 E.2489
G1 X60.718 Y110.016 E.2489
;WIDTH:0.399154
G1 X60.717 Y110 E.00045
G1 X60.718 Y109.985 E.00045
;WIDTH:0.398478
G1 X61.469 Y101.427 E.2489
G1 X63.695 Y93.13 E.2489
```

Note that for the two moves after `WIDTH:0.399154` have distances of less than 0.02 mm. Also note that the width is only about 0.005 mm changed!

I have observed this behavior on PrusaSlicer, but I don't observe it on Cura, so perhaps it's an Arachne issue that has been fixed already in Cura, or perhaps it's a configuration difference.

I think a simple solution might be to allow nominal extrusion width to change by some epsilon amount (perhaps 0.01 mm) without requiring a new segment.

### Expected results

Output where each G1 segment is of a non-trivial length, avoiding movement stalls.

### Project file & Debug log uploads

[Sphere-Shell.3mf.zip](https://github.com/SoftFever/OrcaSlicer/files/13432610/Sphere-Shell.3mf.zip)

### Checklist of files to include

- [X] Log file
- [X] Project file

sethml, 2023-12-01:
Note: I just submitted a pull request to fix this issue upstream in PrusaSlicer: https://github.com/prusa3d/PrusaSlicer/pull/11811

Should I submit a duplicate pull request here, or wait for OrcaSlicer to roll in upstream changes?

SoftFever, 2023-12-06:
This fix looks very promising.
I'll do code review and some testing soon.
Thanks for the PR 👍

closed 2023-12-15 by SoftFever