OrcaSlicer/OrcaSlicer #2841: flow ratio global process

StephInoP, 2023-11-22:
i think you have missed the flow ratio in the global process, usually it's there

i dont know if its a bug or you just forget

and there is one more shown in the pics attached
[image]
[image]

igiannakas, 2023-11-23:
it has been removed as in the global process flow ratio is not necessary. You can achieve the same by setting the filament flow ratio.

StephInoP, 2023-11-23:
how about the flow ratio under process objects? 

my opinion, i think it's better to have ONLY ONE flow ratio which is located at the filament setting, to avoid confusion,
(in previous versions there were 3, one under global, one under objects, one under filament setting) now there are 2,

and can keep bridge flow ratio, bottom/top surface flow ratio as they are now

igiannakas, 2023-11-23:
The object one is useful for things like calibrations. You set your filament to 1 then create 4 cubes with 1, 0.99, 0.98, 0.97 and find the best looking one.

StephInoP, 2023-11-23:
let's say I pick 0.96, then I put the same number in the flow ratio (filament setting and object)?

sorry to disturb you with my curiosity

igiannakas, 2023-11-23:
> let's say I pick 0.96, then I put the same number in the flow ratio (filament setting and object)?
> 
> sorry to disturb you with my curiosity

Then it will use 0.96*0.96 = 0.9216 as the effective rate.

StephInoP, 2023-11-23:
> > let's say I pick 0.96, then I put the same number in the flow ratio (filament setting and object)?
> > sorry to disturb you with my curiosity
> 
> Then it will use 0.96*0.96 = 0.9216 as the effective rate.

aha i get it now, thank you very very much

discip, 2023-11-23:
@StephInoP
Judging by your last response this seems to be solved.
If this is the case, please don't forget to close. 😊

thanks

discip, 2023-11-23:
@StephInoP
Judging by your last response, this seems to be solved.
If this is the case, please don't forget to close. 😊

thanks

closed 2023-11-24 by StephInoP